espressif/idf-eclipse-plugin #779: ESP IDF 5.1 install error

rusiradd, 2023-06-06:
Traceback (most recent call last):
  File "C:\Espressif\esp-idf-v5.1-rc1\tools\idf_tools.py", line 2660, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\esp-idf-v5.1-rc1\tools\idf_tools.py", line 2652, in main
    action_func(args)
  File "C:\Espressif\esp-idf-v5.1-rc1\tools\idf_tools.py", line 1598, in action_export
    tools_info = load_tools_info()
  File "C:\Espressif\esp-idf-v5.1-rc1\tools\idf_tools.py", line 1269, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:  # type: ignore
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Espressif\\tools/tools.json'

This error happens when i try to open a example IDF project

rusiradd, 2023-06-06:
[image]

sigmaaa, 2023-06-06:
Hi @rusiradd 

thank you for reporting this. So far I can't reproduce the issue on Mac and Windows. I've noticed that the command for install tools looks suspicious in your case:
"Executing **python** .... install all" instead of **python** there should be an executable python path like this:
`Executing C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\python3.9.exe D:\work\esp-idf-v5.1-rc1\tools\idf_tools.py install all`

It looks like the plugin was not able to find Python 3.6 and higher on your system (in the Windows registry). Do you have a python from the official vendor https://www.python.org/downloads/?

kolipakakondal, 2023-06-06:
Hi @rusiradd Looks like IDF_PATH is not interpreted correctly. Instead of `C:\Espressif\esp-idf-v5.1-rc1\tools\tools.json` it was looking at `C:\Espressif\tools/tools.json`

kolipakakondal, 2023-06-06:
Hi @rusiradd Could you share the product information. Please post the entire console output.

> Navigate to Espressif > Product Information

dobairoland, 2023-10-30:
@kolipakakondal @alirana01 I've checked the `idf_tools.py` script and patching it to ignore `IDF_PATH` would cause breaking changes.

Would it be possible to you to set the correct IDF_PATH before running `idf_tools.py` or pass it each time with the `--idf-path` command line argument?

kolipakakondal, 2023-10-30:
Hi @dobairoland As a quick solution, we are planning to introduce validation while capturing IDF_PATH in the UI and that should avoid these sort of issues. 

Hi @rusiradd Can you confirm what is the IDF_PATH set in the IDE while installing tools?
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-view-product-information

babelmandeb124, 2024-09-28:
same problem, v1.3.5