espressif/idf-eclipse-plugin #787: Launching Project Encountered Problem "Index 1 out of bound for length 1"

radhikaalpha, 2023-06-19:
I have installed eclipse ide. The versions of all related installed software are as follow,
                      Operating System: windows 10
            Java Runtime Version: 17.0.7+8-LTS-224
            Eclipse Version: 4.27.0.v20230302-0300
            Eclipse CDT Version: 11.1.1.202303201356
            IDF Eclipse Plugin Version: 2.10.0.202306060331

Created a project using existing example code esp_http_client for ESP32-S3-DevKitM-1.
It got built successfully. But it is giving error while flashing the code on Dev Board as 'Launching Project Encountered Problem "Index 1 out of bound for length 1"'
[image]

Can you please guide us for resolving this issue? Are we missing something?
For your information, we have built generic_gpio example and also flashed the same to development kit. So, development kit is working.

kolipakakondal, 2023-06-19:
Hi @radhikaalpha It seems to be LaunchBar is missing in your eclipse cdt. Could you share how did go about installing Eclipse CDT and plugin?

Please try the following.

1. Go to `Install New Software`
2. Add this update site `https://download.eclipse.org/tools/cdt/releases/11.1`
3. Install the Launchbar plugin if it's missing. Please find the screenshot below.

[image]

if it's already installed, for some reason Launchbar is disabled, please enable it
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-enable-the-launchbar-in-espressif-ide

radhikaalpha, 2023-06-20:
Hi, Thanks for your reply. It got resolved by enabling LaunchBar in preferences.

kolipakakondal, 2023-06-20:
Hi @radhikaalpha Thanks for the update and glad that it's working for you. Would you mind sharing how did you install eclipse cdt and plugin? Because if someone installed Eclipse CDT 2023-03 launchbar is enabled by default so I wonder how come it was missing in your case.

radhikaalpha, 2023-06-20:
- We have followed below links to install the IDE
            Install Java 17, Git & python from
                  https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md
            For eclipse installation refer below
                  https://www.beyondlogic.org/getting-started-with-the-espressif-esp-idf-eclipse-ide/
            ESP-IDF with Eclipse
                  https://www.youtube.com/watch?v=OSm-NN97woM

radhikaalpha, 2023-06-20:
When I installedthe IDE, The LaunchBar was there in my IDE. But I don't have any idea where and how it went later. I haven't done anything related to it (ie. haven't manually disabled it)

closed 2023-06-20 by radhikaalpha

kolipakakondal, 2023-06-21:
Ok, thanks for the info.