espressif/idf-eclipse-plugin #794: No `Eclipse Marketplace`  menu entry at Help menu

tonyho, 2023-07-17:
**Describe the bug**
No `Eclipse Marketplace`  menu entry at Help menu.

**Expected behavior**
Open Eclipse marketplace for user to install extensions.

kolipakakondal, 2023-07-18:
Hi @tonyho Is this what you're looking for? It's part of every eclipse installation.

[image]

You can also find esp-idf eclipse plugin in the marketplace
[image]

kolipakakondal, 2023-07-18:
Could you please share your environment details to check further on this

tonyho, 2023-07-18:
> I realized it's missing in the Espressif-IDE. Here is the PR #796↳

Thanks for the fast response.  That's exactly what I mean.

kolipakakondal, 2023-09-22:
Fix is part of the v2.11.0 release. https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.11.0

Hence closing this issue

closed 2023-09-22 by kolipakakondal