espressif/idf-eclipse-plugin #820: Installation of Espressif IDF Plugin on Eclipse CDT 2023-09 (4.29.0) fails

elfrances, 2023-09-15:
I'm setting up a new Ubuntu 22.04.3 PC to develop software for the ESP32 using Eclipse CDT version 2023-09.  Everything goes fine until I try to install the Espressif IDF Plugin following the instructions in the README.md file.  The installer fails with the following error:

Cannot complete the install because one or more required items could not be found.
  Software being installed: Espressif IDF Plugins for Eclipse 2.10.0.202306060331 (com.espressif.idf.feature.feature.group 2.10.0.202306060331)
  Missing requirement: ESP-IDF Serial Monitor 1.0.1.202306060331 (com.espressif.idf.terminal.connector.serial 1.0.1.202306060331) requires 'osgi.bundle; org.slf4j.api 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.10.0.202306060331 (com.espressif.idf.feature.feature.group 2.10.0.202306060331)
    To: org.eclipse.equinox.p2.iu; com.espressif.idf.terminal.connector.serial [1.0.1.202306060331,1.0.1.202306060331]

I have also attached a couple of screenshots and the error log file.

[EclipseCDT-Error-Log.log](https://github.com/espressif/idf-eclipse-plugin/files/12615369/EclipseCDT-Error-Log.log)

[image]

[image]

Thanks in advance for any help you could offer me.

kolipakakondal, 2023-09-15:
Hi @elfrances Thanks for reporting. Let us check and get back to you.

baylf2000, 2023-09-18:
> Hi @elfrances Thanks for reporting. Let us check and get back to you. Meanwhile, you can use [Eclipse CDT 2023-06](https://www.eclipse.org/downloads/packages/release/2023-06/r/eclipse-ide-cc-developers) version and which works fine with the our [plugin](https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/).

I can confirm this error occurs with the 2023-09 version.

w4mmp-ron, 2023-09-19:
Hello,
I am experiencing the same issue only on windows:
Your original request has been modified.
  "Embedded C/C++ GDB JTAG Debugging" will be ignored because a newer version is already installed. 
  "Embedded C/C++ Core" will be ignored because a newer version is already installed. 
  "Embedded C/C++ CMSIS Packs Core Plug-in" will be ignored because a newer version is already installed. 
  "Embedded C/C++ Templates Core Plug-in" will be ignored because a newer version is already installed. 
Cannot complete the install because one or more required items could not be found.
  Software being installed: Espressif IDF Plugins for Eclipse 2.10.0.202306060331 (com.espressif.idf.feature.feature.group 2.10.0.202306060331)
  Missing requirement: ESP-IDF Serial Monitor 1.0.1.202306060331 (com.espressif.idf.terminal.connector.serial 1.0.1.202306060331) requires 'osgi.bundle; org.slf4j.api 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.10.0.202306060331 (com.espressif.idf.feature.feature.group 2.10.0.202306060331)
    To: org.eclipse.equinox.p2.iu; com.espressif.idf.terminal.connector.serial [1.0.1.202306060331,1.0.1.202306060331]
[image]

kolipakakondal, 2023-09-19:
Hi All, Eclipse 2023-09 is not yet supported by our plugin. We are working on it. Hence, please continue to use [Eclipse CDT 2023-06](https://www.eclipse.org/downloads/packages/release/2023-06/r/eclipse-ide-cc-developers) version or earlier.

kolipakakondal, 2023-09-21:
Hi @elfrances @baylf2000  @w4mmp-ron Could you please confirm if you're able to install this [update site](https://github.com/espressif/idf-eclipse-plugin/suites/16356151538/artifacts/934280246) on the Eclipse 2023-09

How to install this build
https://github.com/espressif/idf-eclipse-plugin#installing-idf-eclipse-plugin-from-local-archive

closed 2023-09-21 by kolipakakondal

w4mmp-ron, 2023-09-21:
Hello,
Thank you I was able to install the plugin on Eclipse 2023-09.  
Regards,
Ron

kolipakakondal, 2023-09-22:
We have also released v2.11.0 on 21st sept 2023, this fix is part of the release
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.11.0