espressif/idf-eclipse-plugin #821: Could not find CMAKE_PROJECT_NAME in cache

Mendiz-Fern, 2023-09-18:
I originally started having the same issue as #22, and as I got further and further along their solution, I realized that I was having issues with the build tools, I started debugging and I got to the point where this is what I have in my CMake preferences:
[image]

and this is what I have in the project build target settings:
[image]

As you can see from the second screenshot, I have the cmake file in the CMake preferences, but not on the build target settings, in fact, it's not even an option for me to select. I've already run the install tools multiple times as well as made sure the environment variables were properly set
[image]

I think my core toolchain should be fine:
[image]

I'm not sure if there could be an issue regarding the fact that there are some path discrepancies in there between the core toolchain and the Env. Variables, such as having C:/Espressif or C:/esp-idf-master (I had a very rough time installing IDF tools to begin with, so I'm not even sure where anything is and I'm not entirely sure how to delete everything pertaining to the ESP IDF to start again)

Another fun thing is that when I look at the product info, I get a fatal error: not a git repository, which I don't know why, if I installed it from the ESP-IDF installer and I thought it set it up as a Git. Does it not if it's the offline installer? Idr if I kept the offline or online install.

Product Information:
Operating System: windows 10
Java Runtime Version: 20.0.2+9-78
Eclipse Version: 4.28.0.v20230605-0440
Eclipse CDT Version: 11.2.0.202303140100
IDF Eclipse Plugin Version: 2.10.0.202306060331
ESP-IDF v5.1.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.2

Also there's two error logs so I'm just putting both of them here if this would even help
[novaluepresent.log](https://github.com/espressif/idf-eclipse-plugin/files/12653417/novaluepresent.log)
[filenotfound.log](https://github.com/espressif/idf-eclipse-plugin/files/12653419/errorlog1.log)

kolipakakondal, 2023-09-19:
Hi @Mendiz-Fern I see you are trying to deal with multiple esp-idf environments and somewhere the paths were messed up. 

For example, I see idf_tools.py multiple places.

```
C:\Espressif\tools\idf_tools.py
C:\Espressif\tools\cmake\esp-idf-v5.1.1\tools\idf_tools.py
C:\esp-idf-master\tools\idf_tools.py
```
But in the screenshot you've provided, it was pointing to `C:\esp-idf-master` as a the ESP-IDF path.

I would highly recommend, you go with the offline installer approach, which will bring all the required tools and dependenecies - esp-idf, tools and espressif- ide.

Please check this [video](https://youtu.be/gU3UHvOyvLE?si=HCW6rh73zQzz1fFn) and see if it's helpful.

[Offline installer for espressif-ide](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/Espressif-IDE-Windows-Installer.md) can be downloaded from [here](https://dl.espressif.com/dl/esp-idf/)

[image]

Let me know how it goes, I will be happy to help you.

sigmaaa, 2023-09-19:
Hi @Mendiz-Fern,
thank you for reporting this. While we are in the process of investigating this problem, I would like to offer some clarification on the points you mentioned:

> I have the cmake file in the CMake preferences, but not on the build target settings, in fact, it's not even an option for me to select.
 
you don't need to select cmake file in build settings since it's bind to the toolchain in the CMake preferences. The process of binding the toolchain to the cmake file is automatic.

> I've already run the install tools multiple times as well as made sure the environment variables were properly set

I agree that the environment variables looks totally fine. What doesn't look right is that according to the PATH variable the toolchains should be located in the path `C:\Epressif\tools`, but in the Core Build Toolchains and CMake sections they are mentioned in another path `C:\Users\Weapo\.rustup\toolchains`. I think that's the root of the problem here. Normally, after installing the tools in the IDE, all toolchains should point to the path mentioned in the environment variables. 

> I'm not sure if there could be an issue regarding the fact that there are some path discrepancies in there between the core toolchain and the Env. Variables, such as having C:/Espressif or C:/esp-idf-master 

It's absolutely fine to have toolchains in a separate folder from the esp-idf. It's how it supposed to be. For example I have toolchains located by default at: C:\Users\Denys\.espressif\tools and at the same time I'm using multiple esp-idf versions, which are located like this D:\work\esp-idf-v5.1-rc1. Also, it's fine that cmake files are located in the esp-idf/tools/cmake folder, while toolchains are located at .espressif\tools. By looking [filenotfound.log](https://github.com/espressif/idf-eclipse-plugin/files/12653419/errorlog1.log) I can see that something messed up, because of this message:

> !MESSAGE C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1520.0_x64__qbz5n2kfra8p0\python3.11.exe: can't open file 'C:\\Espressif\\tools\\idf_tools.py': [Errno 2] No such file or directory

idf_tools.py should be searched inside the esp-idf folder, not the toolchains folder

> I had a very rough time installing IDF tools to begin with, so I'm not even sure where anything is and I'm not entirely sure how to delete everything pertaining to the ESP IDF to start again

installing IDF tools not should be hard at all. I'm using multiple versions of the esp-idf by constantly switching between them and not having such an issue. Have you tried to install esp-idf, by clicking "Download and configure ESP-IDF", then choose the option "Download ESP-IDF". After downloading, the IDE will suggest to install according tools. After that, all should be set up to work with the selected esp-idf. I would suggest trying this approach if offline installer installation doesn't work for you.

Mendiz-Fern, 2023-09-19:
the full PATH environment variable is
C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\python_env\idf5.1_py3.11_env\Scripts;C:\Users\Weapo\esp-idf-v5.1.1\tools;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\dotnet\;C:\Program Files\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\ProgramData\chocolatey\bin;C:\Program Files\Wolfram Research\WolframScript\;C:\Program Files\MATLAB\R2022b\bin;C:\Program Files\Git\cmd;C:\Users\Weapo\.rustup\toolchains\esp\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Users\Weapo\.rustup\toolchains\esp\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Users\Weapo\.rustup\toolchains\esp\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Users\Weapo\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-16.0.0-20230516\esp-clang\bin;C:\Users\Weapo\.rustup\toolchains\esp\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\idf-python\3.11.2\;C:\Espressif\tools\idf-git\2.39.2\cmd\;C:\Program Files (x86)\Microsoft\Edge\Application;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\dotnet\;C:\Program Files\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\ProgramData\chocolatey\bin;C:\Program Files\Wolfram Research\WolframScript\;C:\Program Files\MATLAB\R2022b\bin;C:\Program Files\Git\cmd;C:\Tcl\bin;C:\Users\Weapo\AppData\Local\Microsoft\WindowsApps;;C:\Users\Weapo\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\dotnet\;C:\Program Files\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\Common Files\Autodesk Shared\Advance\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\ProgramData\chocolatey\bin;C:\Program Files\Wolfram Research\WolframScript\;C:\Program Files\MATLAB\R2022b\bin;C:\Program Files\Git\cmd;C:\Users\Weapo\.cargo\bin;C:\Users\Weapo\AppData\Local\Programs\jaguar;

which... ... seems a little long (is it supposed to have that many paths in it?). Contains multiple directories , and I'm not entirely sure what's up with that. I also downloaded ESP-IDF from eclipse into C:\\users\Weapo, which I'm not sure if it was the right move, but trying to do it directly in C:\\ gave me an error saying it wasn't allowed to put stuff there.

When you say that the issue is the discrepancy between PATH and the core build toolchains/ CMake, are you suggesting I change the Toolchains and CMake paths, or PATH? I'm having an issue where  C:\\Espressif\tools\etc. doesn't actually have any CMake files listed.