espressif/idf-eclipse-plugin #825: New Espressif-IDE 2.1.1.0 windows install fails after each system restart (IDFGH-11127)

baylf2000, 2023-09-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I installed the Espressif IDE along with the latest (v5.1.1) esp-idf, exactly following the installation instructions. I was able to build and flash various zigbee-related firmwares (for the ESP32C6), including after restarting the IDE many times. All good.

However, as soon as I restart the entire system however, the IDE stops functioning. I do the following:

1. Exit the IDE.
2. Click the Windows restart button.
3. Wait for the system to finish restarting.
4. Start the IDE again.

As soon as I click "build", I get the single error "Build not configured correctly".

As per the instructions I attempted to "Install Tools" again, however this also fails with the following message:
[image]

If I change the launch target to an esp32s3, it asks me confirm deleting the build directory, which I confirm. It will then at least start the build process, but fails due to zigbee issues. However as soon as I change the launch target back to esp32c6, the build refuses to start again with the error message "No toolchain file found". If I try and install the tools again, it fails again with the same error as above. Here is the complete output from the "install tools" option.

```
Installing tools...
This can take a while. Please be patient.
Executing C:\Python311\python.exe H:\Espressif\esp-idf-v5.1.1\tools\idf_tools.py install all
Current system platform: win64
Selected targets are: esp32s3, esp32h2, esp32c6, esp32s2, esp32c3, esp32c2, esp32
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20221002 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20221002 (already installed)
Skipping xtensa-esp32-elf@esp-12.2.0_20230208 (already installed)
Skipping xtensa-esp32s2-elf@esp-12.2.0_20230208 (already installed)
Skipping xtensa-esp32s3-elf@esp-12.2.0_20230208 (already installed)
Skipping esp-clang@15.0.0-23786128ae (already installed)
Skipping riscv32-esp-elf@esp-12.2.0_20230208 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
WARNING: tool cmake version 3.16.3 is installed, but has reported version 3.24.0
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230419 (already installed)
Skipping ninja@1.10.2 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.8 (already installed)
Skipping dfu-util@0.11 (already installed)
Skipping esp-rom-elfs@20230320 (already installed)
OK

Executing C:\Python311\python.exe H:\Espressif\esp-idf-v5.1.1\tools\idf_tools.py install-python-env
Python 3.11.4
pip 23.2.1 from C:\Users\lachl\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\pip (python 3.11)

Upgrading pip and setuptools...
Requirement already satisfied: pip in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (23.2.1)
Requirement already satisfied: setuptools in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (68.2.2)
Skipping the download of C:\Users\lachl\.espressif\espidf.constraints.v5.1.txt because it was downloaded recently.
Installing Python packages
 Constraint file: C:\Users\lachl\.espressif\espidf.constraints.v5.1.txt
 Requirement files:
  - H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 3)) (68.2.2)
Requirement already satisfied: packaging in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 4)) (23.1)
Requirement already satisfied: click in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 7)) (8.0.4)
Requirement already satisfied: pyserial in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 8)) (3.5)
Requirement already satisfied: cryptography in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 9)) (39.0.2)
Requirement already satisfied: pyparsing in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 10)) (3.0.9)
Requirement already satisfied: pyelftools in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from -r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 11)) (0.29)
[... 23 lines cut ...]
Requirement already satisfied: reedsolo<1.8,>=1.5.3 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from esptool->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 14)) (1.7.0)
Requirement already satisfied: kconfiglib>=13.7.1 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from esp-idf-kconfig->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 15)) (14.1.0)
Requirement already satisfied: esp-idf-panic-decoder in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from esp-idf-monitor->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 16)) (0.1.0)
Requirement already satisfied: bitarray<3.0.0,>=2.8.0 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from bitstring>=3.1.6->esptool->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 14)) (2.8.1)
Requirement already satisfied: msgpack>=0.5.2 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6->idf-component-manager->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 12)) (1.0.6)
Requirement already satisfied: filelock>=3.8.0 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6->idf-component-manager->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 12)) (3.12.4)
Requirement already satisfied: pycparser in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from cffi>=1.12->cryptography->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 9)) (2.21)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from requests<3->idf-component-manager->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 12)) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from requests<3->idf-component-manager->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 12)) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\lachl\.espressif\python_env\idf5.1_py3.11_env\lib\site-packages (from requests<3->idf-component-manager->-r H:\Espressif\esp-idf-v5.1.1\tools\requirements\requirements.core.txt (line 12)) (2023.7.22)
OK

Executing C:\Python311\python.exe H:\Espressif\esp-idf-v5.1.1\tools\idf_tools.py export --format=key-value
Not using an unsupported version of tool esp-clang found in PATH: unknown. To use it, run 'C:\Python311\python.exe H:\Espressif\esp-idf-v5.1.1\tools\idf_tools.py export --prefer-system'
WARNING: tool cmake version 3.16.3 is installed, but has reported version 3.24.0
[... cut ...] .espressif\tools\ninja\1.10.2\;C:\Users\lachl\.espressif\tools\idf-exe\1.0.3\;C:\Users\lachl\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\lachl\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\lachl\.espressif\python_env\idf5.1_py3.11_env\Scripts;H:\Espressif\esp-idf-v5.1.1\tools;%PATH%
IDF_DEACTIVATE_FILE_PATH=C:\Users\lachl\AppData\Local\Temp\tmpyrnkov36idf_20944
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
```

sigmaaa, 2023-09-25:
Hi @baylf2000,

thank you for reporting this, unfortunately, we can't reproduce this problem for now, so it's harder to provide a quick fix for it.

> As per the instructions I attempted to "Install Tools" again, however this also fails with the following message:
[image]

Could you provide the error log after this error occurred? https://github.com/espressif/idf-eclipse-plugin#error-log
This will help us better understand the root of the exception

M-Bab, 2024-01-03:
Hi @baylf2000 

I think I got a similar issue but in my case building with eclipse is actually never working for my esp32.

But I discovered a weird workaround: If I directly hit "Debug" (in the launch bar) without selecting build first, it builds and tries to debug/flash. This is not really a handy workaround but maybe you can confirm the behavior in your issue is similar?

But the build (either Launchbar or directly via project menu) does not work. It seams that the Eclipse IDF plugin somehow fails to configure some parts of the projects here (Toolchain, IDF?).

M-Bab, 2024-03-04:
Yeah I can confirm that both versions (2.11 and 2.12) are broken in my use case with the very same errors. Overall it is "toolchain is NULL" in different descriptions. Processor in use is the ESP32 and toolchain is gcc (minimum install by `install.sh esp32`).

My workaround was to uninstall the ESP-IDF Plugin and rollback to the version 2.10:
1. Download the zip file from https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-2.10.0.zip
2. Install the zip addon manually by "Help->Install New Software" then "Add->Archive ..." and select the downloaded archive.
3. Select "Espressif IDF" and its subcomponents and click through the install process

In my case it was working directly without any configuration about IDF paths or new project build/debug configurations.

With 2.10 (also with 2.9.1) everything worked again which looks like a big regression in 2.11/2.12. I think the main reason there are only a few complaints is that a lot of people already switched to VS Code.

Compare this issue: https://esp32.com/viewtopic.php?t=13852