espressif/idf-eclipse-plugin #831: SDK config editor incorrectly handle given value for partition table offset, as HEX without 0x...

Maksons, 2023-10-04:
**Describe the bug**
SDK config editor incorrectly handle given value for partition table offset, as HEX without 0x... 

**To Reproduce**
1) Open SDK Config of project
2) Go to partition table
3) "Offset of partition table" has some !! decimal value !!!
4) Set "Offset of partition table" to 10000 (any other decimal value have problem too)
5) Press save ( Ctrl-S )
6) See wrong value (65536) in the same field ..

[image]

 Ctrl-S
 
[image]

Espressif-IDE V2.11.0 
Linux Mint

igrr, 2023-10-04:
Can confirm, just recently ran into this issue as well.

alirana01, 2023-11-27:
The issue is being worked on and will be fixed with the new release.
https://github.com/espressif/idf-eclipse-plugin/pull/859

alirana01, 2023-12-14:
This PR is merged and closed, the issue will be fixed with latest release I am closing it now. Feel free to open it if you run into this issue after the upcoming release

closed 2023-12-14 by alirana01