espressif/idf-eclipse-plugin #837: ESP-IDF Serial Monitor not starts

userbogd, 2023-10-18:
**Describe the bug**
ESP-IDF Serial Monitor not starts

**To Reproduce**
Steps to reproduce the behavior:
It's very easy to reproduce - simply try to run it!

**Expected behavior**
Showing in header Connecting... but nothing inside the window.
From console runs normally. Serial Terminal from Eclipse also runs normally. 
Every start of monitor accompanies error log. 

**Screenshots**
[image]

**Espressif-IDE Product Information:**
[prod_info.txt](https://github.com/espressif/idf-eclipse-plugin/files/13021551/prod_info.txt)

**Eclipse Error log:** 
[error_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/13021656/error_log.txt)

This issue checked on two different computer with clean eclipse installation.  
Now i rolled back to 2.10 release, i works good. Before this, i stayed on nightly channel and untill begin of septembre all was fine. 
Also i very supprised that only i faced with this problem, maybe this is my problem  only because of any kind of specific configuration or so. I would be very thankfull if somebody check this issue and reply to me! 
Thank you very much for this great product! It's developing and improving are life-critical-important for all esp developers!

sigmaaa, 2023-10-19:
Hi @userbogd,

Thank you for reporting this. The problem happens because of some changes in dependencies. We are already working on a fix and it will be available in the 2.11.1 release. For now, as a workaround, you can call 'idf.py monitor --no-reset -p PORT' command from ESP-IDF Terminal inside the IDE.

kolipakakondal, 2023-10-19:
Hi @userbogd Could you please check if you are able to use serial monitor with this [build](https://github.com/espressif/idf-eclipse-plugin/actions/runs/6576227258?pr=840#artifacts) 
Update site https://github.com/espressif/idf-eclipse-plugin/suites/17411607211/artifacts/995150326
espressif-ide-linux https://github.com/espressif/idf-eclipse-plugin/suites/17411607211/artifacts/995150328

wiegleyj, 2023-10-19:
Stupid question... How do install (or update) that build instead of the marketplace build?

userbogd, 2023-10-19:
> Stupid question... How do install (or update) that build instead of the marketplace build?

It's better first to remove plugin installed from marketplace. After that Help -> Install new software... ->  Add... -> Archive... 
I installed this build without uninstalled 2.11.0, so possible needed first install 2.11.0. You can try to do this if clean install will not helps.

wiegleyj, 2023-10-19:
Thanks, I'll give that a go.

wiegleyj, 2023-10-19:
I can't get it to work. I get: No software site found at jar:file:/C:/Users/jeffw/Downloads/Espressif-IDE-2.11.1-win32.win32.x86_64.zip!/.  (I've also tried the top level zip that was downloaded initially from the site you linked, same error, different file text in the error.)

Unpacking that zip file doesn't help as the archive only allows .zip/.jar.

kolipakakondal, 2023-10-31:
Hello, We have released IEP v2.11.1 with this fix. 
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.11.1

closed 2023-10-31 by kolipakakondal