espressif/idf-eclipse-plugin #839: Basic new project won't compile because "true" is undefined. (because __cplusplus is?)

wiegleyj, 2023-10-18:
**Describe the bug**
Trying to switch to eclipse and I've got the esp-idf eclipse plugin and tools installed. I created a new project with File->New->Espressif IDF Project.

The main.c program shows "while (true) {" as having the error "Symbol 'true' could not be found".

'true' is defined in stdbool.h and stdbool.h is included in main.c.

It appears that somehow main.c is being treated as a c++ file and __cplusplus is defined and thus cutting out all the boolean defines that are intended for C files. I don't think __cplusplus should be defined for *.c files. I can't find where there is definition is taking place.

**To Reproduce**
Create a new Espressif IDF project.  File->New->Espressif IDF Project
Try to build.

**Expected behavior**
main.c is expected to build cleanly.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Executing C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1776.0_x64__qbz5n2kfra8p0\python3.11.exe C:\Espressif\frameworks\esp-idf-v5.1.1\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20221002 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20221002 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20230419 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.8 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.11 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
OK

CDT Build environment variables
PATH: C:\Users\jeffw\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Users\jeffw\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Users\jeffw\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Users\jeffw\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Users\jeffw\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Users\jeffw\.espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Users\jeffw\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Users\jeffw\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\jeffw\.espressif\tools\cmake\3.24.0\bin;C:\Users\jeffw\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\Users\jeffw\.espressif\tools\ninja\1.10.2\;C:\Users\jeffw\.espressif\tools\idf-exe\1.0.3\;C:\Users\jeffw\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\jeffw\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\jeffw\.espressif\python_env\idf5.1_py3.11_env\Scripts;C:\Espressif\frameworks\esp-idf-v5.1.1\tools;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Docker\Docker\resources\bin;C:\Users\jeffw\AppData\Local\Microsoft\WindowsApps;C:\Users\jeffw\AppData\Local\GitHubDesktop\bin;C:\Users\jeffw\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:\Users\jeffw\.dotnet\tools;C:\Users\jeffw\AppData\Local\Programs\Microsoft VS Code\bin
USERDOMAIN_ROAMINGPROFILE: DESKTOP-UB7S1L9
PROCESSOR_LEVEL: 6
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
ESP_IDF_VERSION: 5.1
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
OPENOCD_SCRIPTS: C:\Users\jeffw\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts
IDF_DEACTIVATE_FILE_PATH: C:\Users\jeffw\AppData\Local\Temp\tmpehboaqccidf_16264
IDF_CCACHE_ENABLE: 1
USERNAME: jeffw
ProgramFiles(x86): C:\Program Files (x86)
FPS_BROWSER_USER_PROFILE_STRING: Default
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
OneDriveConsumer: C:\Users\jeffw\OneDrive
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
HOMEPATH: \Users\jeffw
PROCESSOR_IDENTIFIER: Intel64 Family 6 Model 183 Stepping 1, GenuineIntel
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\Windows
=::: ::\
ZES_ENABLE_SYSMAN: 1
IGCCSVC_DB: AQAAANCMnd8BFdERjHoAwE/Cl+sBAAAA/Qwx+Os0vUyDHuesv+o2iQQAAAACAAAAAAAQZgAAAAEAACAAAACriWE5GvP5QF9KEN4c/y9TkIX3Upw61u3a/7sS85qmgQAAAAAOgAAAAAIAACAAAACXDPHZJYEQeT3G4ongKmajRjGi12Xf01t7MA+p1sCkxWAAAAA4mMKcS6yUNT9L7JLwDJ+Yuw4YrWCGRySc8sg/CYZ2KymE1L+iw5TTb/JpgoXh28RoWtQxvZHwhHRVOjNRscrnfiHxejn7BCR0WjBIU51TOdfbt0KsMWEhS/jI9ZRxq0FAAAAAGysGmQGntgWq/69diRMHcFVyqso5tWkFxScaPkInNMkjABAR6w7R7Q/Kj/8B9tC1OOy+mNgohV1HJ/sHjitb+g==
LOCALAPPDATA: C:\Users\jeffw\AppData\Local
USERDOMAIN: DESKTOP-UB7S1L9
LOGONSERVER: \\DESKTOP-UB7S1L9
FPS_BROWSER_APP_PROFILE_STRING: Internet Explorer
OneDrive: C:\Users\jeffw\OneDrive
APPDATA: C:\Users\jeffw\AppData\Roaming
EFC_9592: 1
CommonProgramFiles: C:\Program Files\Common Files
OS: Windows_NT
COMPUTERNAME: DESKTOP-UB7S1L9
IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.1.1
PROCESSOR_REVISION: b701
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
STM32CubeMX_PATH: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX
IDF_MAINTAINER: 1
TEMP: C:\Users\jeffw\AppData\Local\Temp
SystemRoot: C:\Windows
HOMEDRIVE: C:
USERPROFILE: C:\Users\jeffw
TMP: C:\Users\jeffw\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
CDS_HOME: C:\cds_spb_home
NUMBER_OF_PROCESSORS: 32
IDF_PYTHON_ENV_PATH: C:\Users\jeffw\.espressif\python_env\idf5.1_py3.11_env
ESP_ROM_ELF_DIR: C:\Users\jeffw\.espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 11
Java Runtime Version: 20.0.2+9-78
Eclipse Version: 4.29.0.v20230903-1000
Eclipse CDT Version: 11.3.0.202307110141
IDF Eclipse Plugin Version: 2.11.0.202309211121
ESP-IDF v5.1.1
Python set for IDF_PYTHON_ENV: Python 3.11.6

**Eclipse Error log:** 
[bugtest.log](https://github.com/espressif/idf-eclipse-plugin/files/13031181/bugtest.log)

sigmaaa, 2023-10-19:
Hi @wiegleyj,

I've noticed that you have the Launch Bar disabled. It's important to have it enabled to build the project with correct configuration.  Please follow the below instructions to enable the LaunBar.

- Go to Preferences...
- Navigate to Run/Debug > Launching > LaunchBar
- Check Enable the LaunchBar
- Click Apply and Close

If you still see an unresolved issue after enabling the launch bar and building the project with it, it is likely caused by some C/C++ 20 language features that are not yet supported by the Eclipse CDT indexer. You can ignore them and make the following settings: https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-disable-the-unresolved-symbol-error-markers

kolipakakondal, 2023-10-19:
I just verified with the Eclipse 20230905-1907 build and ESP-IDF 5.1.1, I don't see any issue with the default example.

Having a launchbar and building a specific target should fix the issue

wiegleyj, 2023-10-19:
I enabled launch bar. Thank you, that was helpful (though I don't think it altered anything). The project compiles (I think it did before too). I can flash it and run.

true is still not defined. I'm super skeptical that stdbool.h has changed in C/C++20. I'm not keen on the "solution" to be disabling ALL unresolved symbol errors.

Does it have anything to do with the workspace thinking that *.C (not *.c) are mapped as C++ files? This seems to be an IBM convention and windows doesn't treat filenames as case sensitive. Is it accidentally thinking main.c is main.C and defining __cplusplus?

sigmaaa, 2023-10-24:
Hi @wiegleyj. I think I was able to reproduce the issue or faced a similar issue. In my case, the indexer was not able to resolve some includes until I close/reopen the project and click on build again. While we further investigating this issue, I recommend you try to close/reopen the project and click build. Also,  my indexer settings look like this:
[image]

and project settings:
[image]

> I'm not keen on the "solution" to be disabling ALL unresolved symbol errors.

the other possible solution is to suppress the exact place where this issue happens. To do this. Click on bug marker on the specific line and then "Supress problem "Symbol is not resolved""