espressif/idf-eclipse-plugin #855: Attempted to upgrade the plugin, and it completely uninstalled itself

mbratch, 2023-11-16:
**Describe the bug**
I was running an older Espressif plugin. I executed Help > Check for Updates, then selected only the Espressif plugin to upgrade it.

After the update and Eclipse restarted, there was no Espressif plugin visible, and other Eclipse settings were changed (e.g., my dark theme changed back to default light).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Help > Check for Updates
2. Select only the Espressif plugin
3. Click buttons to install
4. Eclipse requests restart, so restart
5. Espressif plugin no longer installed and other environment settings have defaulted; the Espressif menu is not available

**Expected behavior**
I expected to reboot Eclipse and the new plugin would be installed and my other environment settings unchanged

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)
Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 

(I will add later)

AndriiFilippov, 2023-11-16:
@mbratch, hello ! 

Could you please provide the following information to make the investigation faster: 

- your OS
- your Eclipse version
- IDE version of "I was running an older Espressif plugin" 

If you have any questions, feel free to ask.
Thank you !

mbratch, 2023-11-16:
Hello @AndriiFilippov, thank you for the quick response!

Sorry about the lack of detail in the original post. I didn't have it handy. I updated the description with the requested information.

I also try outright installing the latest plugin as if it were a new install but I got the same results: nothing seemed to install. I had to restore my Eclipse folder from file history to get back to normal.

sigmaaa, 2023-11-17:
Hi @mbratch, 

thank you for reporting the issue. Based on the information you have provided, it is likely that you have only updated the Espressif plugin without the dependencies required for it to work. To update dependencies as well, you need to go to Help -> Install New Software... -> work with: https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
[image]

mbratch, 2023-11-17:
Thank you @sigmaaa .

Since I already had a prior version of the Espressif plugin installed, I followed these instructions:

[image]

Do the instructions you linked need updating? I am not sure how I would glean your recommendation from the instructions. I will try your suggestion and see if it works. But since I already have a prior version installed, do I need to remove it first?

sigmaaa, 2023-11-17:
Hi @mbratch,

thanks for pointing this out, I think this part of the documentation could be improved and we will work on it.
>  But since I already have a prior version installed, do I need to remove it first?

I think that it is not necessary to remove it. After installing/updating the requirements from the "Install New Software..." page and restarting Eclipse, the plugin should work unless an error occurs during installation.

mbratch, 2023-11-17:
Hi @sigmaaa. When I go to Help -> Install New Software... -> work with: https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ I get this:

[image]

sigmaaa, 2023-11-17:
> Hi @sigmaaa. When I go to Help -> Install New Software... -> work with: https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ I get this:
> 
> [image]

Hi @mbratch, that's strange. Could you please provide the error log after Eclipse startup? Instructions here https://github.com/espressif/idf-eclipse-plugin#error-log. Thank you.

mbratch, 2023-11-17:
Hi @sigmaaa . Unfortunately, my Eclipse does not have the Error Log menu:

[image]

kolipakakondal, 2023-11-18:
> Hi @sigmaaa . Unfortunately, my Eclipse does not have the Error Log menu:
> 

Please click on `Other...` and search for `Error Log`.

Can you also check if you're in `C/C++ perspective`, If you are not in C/C++ perspective the related menu items are not visible.

[image]

Please also verify, if the IEP plugins installed correctly

Eclipse > About Eclipse > Click on `Installation Details` from the dialog >  Click on `Features` Tab >  Look for `ESP-IDF Eclipse Plugin` 

[image]

**Note:**
In parallel you can try setting up new environment as it won't take more than 5-10 mins for everything and you should be able to work your existing project.
- Please download latest version of [Espressif-IDE 2.11.1](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.11.1-win32.win32.x86_64.zip)
- Launch with an existing workspace
- Go to `Espressif` Menu > ESP-IDF Tools Manager > Install Tools > 
- Choose your existing IDF directory which you're using earlier, git and python
- Click on `Install Tools` (This won't take much time as tools are already installed earlier, this will set the environement variables and ready the build environment)
- You're good to start with your project again!

Please see that helps!

mbratch, 2023-11-18:
@kolipakakondal thanks for the tip. I was able to find the Error Log under Show View > Other... > General.

It is attached.
[EclipseError.log](https://github.com/espressif/idf-eclipse-plugin/files/13398956/EclipseError.log)

mbratch, 2023-11-18:
@kolipakakondal @sigmaaa I just checked the plugin version installed for Espressif and now it appears to be the latest version already. So the system seems to be working. I'm not sure why I had the original error. Thank you for all the direction. I will consider this issue closed for the time being.

closed 2023-12-09 by mbratch