espressif/idf-eclipse-plugin #857: Open OCD not working after update to latest version of IDF plugin and all attempts to make it work failed

ralphde, 2023-11-20:
We are not getting Open OCD running after update to latest versions of Eclipse and IDF-plugin on brand new Win11 machine. We tried all possible ways of installing. The Problem seem to be USB driver / ESP-prog related.

We need desperate ideas how to fix this. 

In the first attempt we used only the ESP-prog board for JTAG flashing/debug. At least the flashing went through with image verify. BUT the program (in all cases we tried with the HTTP simple server example) did panic on first try and after second try OCD did never start again. Even a reboot did not help. Very strange. 

after installing the USB driver for SiliconLabs USB bridge on a Saola-1 board for test purpose and reverting back to Libusb using Zadig we  have this issue and can't get it back to what it was before, at least flashing the image:

We tried a lot including changing adapter speed and flash speed

```
> Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 20000 kHz
> Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
> Flashing "C:/eclipse-cpp-2023-03-R-workspace/simple/build/bootloader/bootloader.bin" at 0x1000
> ** program_esp input args <0x1000 verify> **
> Error: failed to reset FTDI device: LIBUSB_ERROR_PIPE
> Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
> ** OpenOCD init failed **
> shutdown command invoked
> ** Flashing Failed **
> -1
> Started by GNU MCU Eclipse
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> 
```
We tried to connect a Saola board because we have not been successful connecting JTAG using ESP-Prog before.

We're using our ESP-prog board since a couple of years now. May it be that the firmware on the ESP-prog are outdated?
How can we use older OCD versions in the IDF-plugin environment/where to get them?

Executing C:\Users\ralph\AppData\Local\Programs\Python\Python312\python.exe C:\IDF-5-1-2\esp-idf-v5.1.2\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20221002 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20221002 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20230921 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.8 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.11 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
OK

CDT Build environment variables
PATH: C:\Users\ralph\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Users\ralph\.espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Users\ralph\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Users\ralph\.espressif\tools\cmake\3.24.0\bin;C:\Users\ralph\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\ralph\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Users\ralph\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\ralph\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Users\ralph\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\ralph\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Users\ralph\.espressif\tools\ninja\1.10.2;C:\Users\ralph\.espressif\tools\idf-exe\1.0.3;C:\Users\ralph\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Users\ralph\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Program Files\Git\cmd\git.EXE;C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env\Scripts\;C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env\bin\;C:\IDF-5-1-2\esp-idf-v5.1.2\tools\;C:/eclipse-cpp-2023-03-R/eclipse/plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.9.v20231028-0858/jre/bin/server;C:/eclipse-cpp-2023-03-R/eclipse/plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.9.v20231028-0858/jre/bin;C:/eclipse-cpp-2023-03-R/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.6.v20230204-1729/jre/bin/server;C:/eclipse-cpp-2023-03-R/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.6.v20230204-1729/jre/bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\AutoFirma\AutoFirma;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Git\cmd;C:\Users\ralph\AppData\Local\Programs\Python\Launcher\;C:\Users\ralph\AppData\Local\Microsoft\WindowsApps;;C:\eclipse-cpp-2023-03-R\eclipse;;C:\eclipse-cpp-2023-03-R\eclipse;
USERDOMAIN_ROAMINGPROFILE: RALPH2
PROCESSOR_LEVEL: 25
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
ESP_IDF_VERSION: 5.1
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
VXIPNPPATH: C:\Program Files (x86)\IVI Foundation\VISA\
OPENOCD_SCRIPTS: C:\Users\ralph\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
IDF_DEACTIVATE_FILE_PATH: C:\Users\ralph\AppData\Local\Temp\tmpa51vfdgmidf_20824
IVIROOTDIR64: C:\Program Files\IVI Foundation\IVI\
IDF_CCACHE_ENABLE: 1
USERNAME: ralph
ProgramFiles(x86): C:\Program Files (x86)
VXIPNPPATH64: C:\Program Files\IVI Foundation\VISA\
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
OneDriveConsumer: C:\Users\ralph\OneDrive
NIEXTCCOMPILERSUPP: C:\Program Files (x86)\National Instruments\Shared\ExternalCompilerSupport\C\
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
__PSLockDownPolicy: 0
HOMEPATH: \Users\ralph
PROCESSOR_IDENTIFIER: AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD
IVIROOTDIR32: C:\Program Files (x86)\IVI Foundation\IVI\
PYTHON_EXE_PATH: C:\Users\ralph\AppData\Local\Programs\Python\Python312\python.exe
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\WINDOWS
=::: ::\
LOCALAPPDATA: C:\Users\ralph\AppData\Local
GIT_PATH: C:\Program Files\Git\cmd\git.EXE
USERDOMAIN: RALPH2
LOGONSERVER: \\RALPH2
OneDrive: C:\Users\ralph\OneDrive
APPDATA: C:\Users\ralph\AppData\Roaming
CommonProgramFiles: C:\Program Files\Common Files
OS: Windows_NT
COMPUTERNAME: RALPH2
NIIVIPATH: C:\Program Files (x86)\IVI Foundation\IVI\
IDF_PATH: C:\IDF-5-1-2\esp-idf-v5.1.2
PROCESSOR_REVISION: 5000
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\WINDOWS\system32\cmd.exe
IDF_MAINTAINER: 1
TEMP: C:\Users\ralph\AppData\Local\Temp
SystemRoot: C:\WINDOWS
HOMEDRIVE: C:
USERPROFILE: C:\Users\ralph
TMP: C:\Users\ralph\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 12
IDF_PYTHON_ENV_PATH: C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env
ESP_ROM_ELF_DIR: C:\Users\ralph\.espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 11
Java Runtime Version: 17.0.9+9
Eclipse Version: 4.27.0.v20230302-0300
Eclipse CDT Version: 11.1.0.202212091724
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v5.1.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.12.0

kolipakakondal, 2023-11-21:
Hi @ralphde Can you enable OpenOCD verbose and share the console output to check further?

[image]

For troubleshooting this issue, please check
https://github.com/espressif/openocd-esp32/issues/174
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ

gerekon, 2023-11-23:
@ralphde Could you provide detailed OpnOCD log?
 Also it would be good to try using previous version of OpenOCD, please see here https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md#debugger-tab how to configure path to OpenOCD.

If in your current setup both OpenOCD have the same problem the root cause is the libusb driver configuration. And I'd recommend in that case removing corresponding JTAG USB device competely in device manager, reboot the system and install driver again.

erhankur, 2023-11-23:
I would clean up/remove any related drivers to the initial stage, and then, with the help of the IDF Windows installer, I would re-install all of them properly. (Including FTDI and USB-serial-jtag)