espressif/idf-eclipse-plugin #858: [Espressif IDE/ESPC3] OpenOCD/GDB/debug suddently stop working

ads17, 2023-11-21:
# Problem description
I am working with ESPC3 on Espressif IDE, with USB-builtin-jtag.
Originaly, I was using the **v5.0.2** and there was no problem.
Sudently, debugging became really unreliable; the vast majority of trials concluded with the exact same result as "Starting OpenOCD timed out".
**[Screenshot 1]**

Nevertheless, when it succeed to start the debug session, either Espressif IDE freezes completely (2), either it doesn’t halt at "app_main" breakpoint and have erratic behaviour.
**[Screenshot 2]**

What seems to me the strangest, is that openocd is working as expected onto ESP-IDF powershell:
**[Screenshot 3]**

These are all the attempts I tried (without solving the problem..):

1. Creating a new "template" project (i used the hello world example),
2. Increase the GDB timeout through (set remotetimeout) on Debug configuration,
3. Updating to Espressif v5.1.1 using the [espressif-ide-setup-2.11.0-with-esp-idf-5.1.1.exe](https://dl.espressif.com/dl/idf-installer/espressif-ide-setup-2.11.0-with-esp-idf-5.1.1.exe?),

# To reproduce
On espressif 5.1.1.

# Expected behaviour
Espressif IDE shall be able to start a debug session.

# Screenshot

- **Screenshot 1**
<image src="https://github.com/espressif/idf-eclipse-plugin/assets/73336788/f1860b36-6e90-4ca9-a31a-88e5dcc63891)" width=300>

- **Screenshot 2**
<image src="https://github.com/espressif/idf-eclipse-plugin/assets/73336788/9e8a88c9-ea46-4a94-97b4-3d37717351f9" width="200">

- **Screenshot 3**
<image src="https://github.com/espressif/idf-eclipse-plugin/assets/73336788/a2ab0881-1ac3-465b-98e6-087c376e70b0" width="400">

- **Screenshot 4**
<image src="https://github.com/espressif/idf-eclipse-plugin/assets/73336788/d6ce4efc-7a10-4fee-8217-2b9b02a591fb" width="400">

# Espressif-IDE Product Information
[product_info.txt](https://github.com/espressif/idf-eclipse-plugin/files/13424746/product_info.txt)

# Eclipse Error log
[errors.log](https://github.com/espressif/idf-eclipse-plugin/files/13424753/errors.log)

ads17, 2023-11-28:
Hello @sigmaaa,

Apologies for my delayed response. I was finally able to solve the issue by reverting to the original version of ESP-IDF (5.0.2) and modifying the GDB server launch timeout(s) parameter from 25 to 100 onto (Windows>Preferences>Espressif).

Previously, I was trying to modify the GDB command in the Debug configuration ("set remotetimeout xx") without any apparent effect.

In my case, the problem appeared at using the bt API, which introduced a significant delay in the launch process.
I remain skeptical about upgrading to version 5.1.2 now that I have a functional workspace.

On another note, from what I understood, the debug session initializes after flash + reset. Is there a way to avoid this reset preceding the debug session? I believe that could help me avoid adjusting this timeout parameter.

sigmaaa, 2023-12-05:
> On another note, from what I understood, the debug session initializes after flash + reset. Is there a way to avoid this reset preceding the debug session? I believe that could help me avoid adjusting this timeout parameter.

Hi @ads17,

To achieve this, try to edit these options:
1. Initialization commands (Startup tab when editing the debug configuration):
- disable the initial reset and make a command like this one: 
``` 
mon halt
set remote hardware-watchpoint-limit {IDF_TARGET_CPU_WATCHPOINT_NUM} 
``` 
[image]

2. Disable Pre-run/Restart reset:
[image]

Hope this helps.