espressif/idf-eclipse-plugin #861: Eclipse indexer unable to find `std` or the first C++ include in files

mbratch, 2023-12-05:
I am running Eclipse version 2023-09 (4.29.0) and have the Espressif ESP-IDF plugin version 2.11.1.202310270725 installed.

I installed the clangd indexer and disabled the default indexer per the instructions at [this link](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md). The installation of the clangd indexer went smoothly, but to get it to take effect, I had to delete my project and re-import it.

My first problem with the indexer is that it seems to have trouble finding standard C++ headers. It says "Use of undeclared identifier 'std'" when I reference an `std` library method.

[image]

It also claims it can't find the string header file as indicated below. I noticed that this happens with the first #include in the file. So if I first #include <memory> then #include <string> it will indicate memory cannot be found but it finds string. Weird.

[image]

FInally, I also don't see any context menu commands for "Open declaration". The other indexer supported it. Is that feature unavailable with the clangd indexer?

I've attached the Eclipse log.
[eclipse.log](https://github.com/espressif/idf-eclipse-plugin/files/13553752/eclipse.log)

mbratch, 2023-12-09:
<tap> <tap> <tap>... is this thing on? :D

kolipakakondal, 2023-12-11:
Hi @mbratch This is a clangd thing!. 

I have configured query driver to `/Users/kondal/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/**/xtensa-esp32-elf-*` to resolve this. Probably, we can automate these things going forward.

Please check more details here.
https://releases.llvm.org/10.0.0/tools/clang/tools/extra/docs/clangd/Configuration.html#query-driver

> Clangd makes use of clang behind the scenes, so it might fail to detect your standard library or built-in headers if your project is making use of a custom toolchain. That is quite common in hardware-related projects, especially for the ones making use of gcc (e.g. ARM’s arm-none-eabi-gcc).
> 
> You can specify your driver as a list of globs or full paths, then clangd will execute drivers and fetch necessary include paths to compile your code.
> 
> For example if you have your compilers at:
> /path/to/my-custom/toolchain1/arm-none-eabi-gcc,
> /path/to/my-custom/toolchain2/arm-none-eabi-g++,
> /path/to/my-custom2/toolchain/arm-none-eabi-g++,
> you can provide clangd with –query-driver=/path/to/my-custom/**/arm-none-eabi* to enable execution of any binary that has a name starting with arm-none-eabi and under /path/to/my-custom/. This won’t allow execution of the last compiler.

[image]

kolipakakondal, 2023-12-11:
>FInally, I also don't see any context menu commands for "Open declaration". The other indexer supported it. Is that feature unavailable with the clangd indexer?

Open declaration menu is not available in the editor yet, but you can access this by pressing F3 or Ctrl+Click (I'm on macOS)

https://github.com/espressif/idf-eclipse-plugin/assets/8463287/e1ee211f-8c6f-401a-a3a2-7d24fb02e339

mbratch, 2023-12-13:
Thanks much @kolipakakondal. Unfortunately, under Windows F3 and Ctrl-Click did not do anything.
I raised my issue in the clangd/clangd project on github.

kolipakakondal, 2023-12-14:
Hi @mbratch I have verified now on Windows, it seems to be F3 and Ctrl+Click are working fine for me. However, I see editor syntax colors are not shown, will look into this.

https://github.com/espressif/idf-eclipse-plugin/assets/8463287/1edcc982-01e9-401f-a31a-4ee9aa3a82db

mbratch, 2023-12-15:
Thanks @kolipakakondal. I'm not sure what went wrong the first time I tried it, but the F3 is working for me now.

kolipakakondal, 2023-12-26:
As this can be resolved by setting the query driver path in the CDT LSP preferences, closing this issue. If you see any other issues, please report us back. thank you.

closed 2023-12-26 by kolipakakondal