espressif/idf-eclipse-plugin #864: I really don't understand why everything has to be so hard

mrhempman69, 2023-12-07:
**Describe the bug**
Cannot create new Espressif-IDF project

**To Reproduce**
File, new, create create new Espressif-IDF project. 

**Expected behavior**
What do you think

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)
```
Operating System: windows 10
Java Runtime Version: 17.0.8.1+1
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v5.1.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.9.13
```
**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

```
java.lang.NoClassDefFoundError: org/eclipse/tools/templates/core/IGenerator
	at com.espressif.idf.ui.handlers.NewProjectHandler.execute(NewProjectHandler.java:25)
	at com.espressif.idf.ui.wizard.NewIdfProjectMainWizard$1.run(NewIdfProjectMainWizard.java:34)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:132)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4046)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3662)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:342)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:648)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:342)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:555)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:208)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:402)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:651)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:588)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)
Caused by: java.lang.ClassNotFoundException: org.eclipse.tools.templates.core.IGenerator cannot be found by com.espressif.idf.ui_1.0.1.202310270725
	at org.eclipse.osgi.internal.loader.BundleLoader.generateException(BundleLoader.java:541)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass0(BundleLoader.java:536)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:416)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:168)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
	... 27 more
```

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

mrhempman69, 2023-12-07:
Are Eclipse developers evil? I already have Eclipse 22-09 installed as well, because of another library (bouffalo-sdk) that was broken by later updates from them. 

What if this repo had been abandoned? All this work would just stop working with eclipse permanently? Have they heard of backwards compatibility?

kolipakakondal, 2023-12-07:
Some of these API changes caused an issue https://github.com/eclipse-cdt/cdt/blob/5250f5a07b992458a0395ada47137e534593f276/NewAndNoteworthy/CHANGELOG-API.md#api-changes-in-cdt-114

kolipakakondal, 2023-12-13:
Hi @mrhempman69 We have added a support for Eclipse CDT 2023-12 on our master, here is the [update site build](https://github.com/espressif/idf-eclipse-plugin/suites/19010563585/artifacts/1112027763).

To install:
1. [Download](https://github.com/espressif/idf-eclipse-plugin/suites/19010563585/artifacts/1112027763)
2. Help > Install New Software
3. Click on Add...
4. Click on Archive.. and choose downloaded zip file and give some readable name
5. Click on Select All
6. Click on Finish

iarug, 2023-12-19:
I experienced the same problem @mrhempman69 described, but in Lubuntu. Solved it by installing webkit2gtk. 
`sudo apt install libwebkit2gtk-4.0-37`
I don't know if the same library is available for windows. Hope this helps some other person with the same problem.

pa-ra-kram, 2023-12-21:
> Hi @mrhempman69 We have added a support for Eclipse CDT 2023-12 on our master, here is the [update site [...]

This fixed the issue in Mac M1 air too. I am guessing this patch is not yet released on the official stable channel yet.
Thanks for this.

kolipakakondal, 2023-12-26:
This is fixed with IEP v2.12.0 release, please check the release here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.0

Please update to the latest and let us know if you see any issues.

closed 2023-12-26 by kolipakakondal

jonesman, 2024-06-20:
The same thing happened to me today with Espressif IDE 3.0.0 on Debian Linux 12 with a full KDE Plasma installation (v.5.27.5).
I do get the same exception, but only after finding this issue here through a Google search did I look at the error log. If you don't go out of your way opening the error log, it just appears as if nothing is happening.
Should this be added to the prerequisites in the setup guide ?

Just for the record, I do appear to have GTK 4 installed, and installing this package fixed the problem: libwebkit2gtk-4.1-0
Exception from error log:
org.eclipse.swt.SWTException: Failed to execute runnable (org.eclipse.swt.SWTError: No more handles because there is no underlying browser available.
Please ensure that WebKit with its GTK 3.x/4.x bindings is installed.)

kolipakakondal, 2024-06-21:
Hi @jonesman Could you please attach the error log here to take a look? So you're able to solve this by installing libwebkit2gtk?

jonesman, 2024-06-21:
Hi @kolipakakondal, yes it was fixed by installing libwebkit2gtk-4.0-37 as ghost did above.
Here is the exception from the error log, thank you for getting back to me:

`org.eclipse.swt.SWTException: Failed to execute runnable (org.eclipse.swt.SWTError: No more handles because there is no underlying browser available.
Please ensure that WebKit with its GTK 3.x/4.x bindings is installed.)
	at org.eclipse.swt.SWT.error(SWT.java:4918)
	at org.eclipse.swt.SWT.error(SWT.java:4833)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:135)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:5039)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4519)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:342)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:648)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:342)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:555)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:208)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:402)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:651)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:588)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1432)
Caused by: org.eclipse.swt.SWTError: No more handles because there is no underlying browser available.
Please ensure that WebKit with its GTK 3.x/4.x bindings is installed.
	at org.eclipse.swt.SWT.error(SWT.java:4944)
	at org.eclipse.swt.browser.Browser.<init>(Browser.java:119)
	at com.espressif.idf.ui.templates.FormBrowser.createControl(FormBrowser.java:53)
	at com.espressif.idf.ui.templates.BaseWizardSelectionPage.createDescriptionIn(BaseWizardSelectionPage.java:34)
	at com.espressif.idf.ui.templates.AbstractTemplatesSelectionPage.createControl(AbstractTemplatesSelectionPage.java:146)
	at org.eclipse.jface.wizard.Wizard.createPageControls(Wizard.java:180)
	at org.eclipse.jface.wizard.WizardDialog.createPageControls(WizardDialog.java:744)
	at org.eclipse.jface.wizard.WizardDialog.createContents(WizardDialog.java:636)
	at org.eclipse.jface.window.Window.create(Window.java:431)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1094)
	at com.espressif.idf.ui.handlers.NewProjectHandler.execute(NewProjectHandler.java:30)
	at com.espressif.idf.ui.wizard.NewIdfProjectMainWizard$1.run(NewIdfProjectMainWizard.java:34)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:132)
	... 24 more`

kolipakakondal, 2024-06-23:
https://github.com/eclipse-platform/eclipse.platform.swt/issues/843