espressif/idf-eclipse-plugin #867: No dialog to create an Espressif IDF project after installing esp-idf-v5.1.2 at Eclipse 2023-12

grobi181, 2023-12-09:
Eclipse 2023-12; esp-idf-v5.1.2; Windows 10 Pro 22H2 Buil19045.3693

After selecting File->NEW->Espressif-IDF-Project, no window appears for creating a new project.

I followed the instructions and installed all components as described.
There is an error message in the error log!

Thanks for help!

[Product_Information_IDE.log](https://github.com/espressif/idf-eclipse-plugin/files/13625259/Product_Information_IDE.log)
[Error_logfile.log](https://github.com/espressif/idf-eclipse-plugin/files/13625261/Error_logfile.log)

sigmaaa, 2023-12-09:
Hi @grobi181,

there are some changes in the latest eclipse CDT API which caused this issue https://github.com/eclipse-cdt/cdt/blob/5250f5a07b992458a0395ada47137e534593f276/NewAndNoteworthy/CHANGELOG-API.md#api-changes-in-cdt-114. We will fix it ASAP. While we fixing this, I recommend you use Espressif-IDE which comes with Eclipse 2023-09+ IEP v2.11.1:
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.11.1-win32.win32.x86_64.zip. Thanks

grobi181, 2023-12-10:
Thanks! I will change to the recommend version! Perfect!!

kolipakakondal, 2023-12-13:
Hi @grobi181 We have added a support for Eclipse CDT 2023-12 on our master branch, here is the [update site build](https://github.com/espressif/idf-eclipse-plugin/suites/19010563585/artifacts/1112027763).

To install:

1. [Download](https://github.com/espressif/idf-eclipse-plugin/suites/19010563585/artifacts/1112027763)
2. Help > Install New Software
3. Click on Add...
4. Click on Archive.. and choose downloaded zip file and give some readable name
5. Click on Select All
6. Click on Finish

RobertGnz, 2023-12-19:
> Hi @grobi181 We have added a support for Eclipse CDT 2023-12 on our master branch, here is the [update site [...]

This does not work. Please TEST your solutions before giving them or don't give any. Thanks.

RobertGnz, 2023-12-19:
> Hi @grobi181 We have added a support for Eclipse CDT 2023-12 on our master branch, here is the [update site [...]

Does not work.

kolipakakondal, 2023-12-26:
This is fixed with IEP v2.12.0 release, please check the release here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.0

Please update to the latest and let us know if you see any issues.

closed 2023-12-26 by kolipakakondal

mbratch, 2024-02-24:
> This is fixed with IEP v2.12.0 release, please check the release here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.0
> 
> Please update to the latest and let us know if you see any issues.

When I install this version, all of my Espressif specific commands in the IDE disappear, including the "Espressif" menu and all the ESP-IDF context menus.

sigmaaa, 2024-02-27:
Hi @mbratch,

what version of Eclipse and versions of eclipse cdt, eclipse embedded CDT plugins are you currently using? To check that, open Help -> About Eclipse IDE ->then you will see a dialog like this:
[image]

Click on highlighted icons to see the versions of the plugins like this:
[image]

The issue probably happened due to dependencies version mismatch. I think you should try to update all the dependencies plugins to the latest versions, which should work fine with our latest plugin. To do that, click Help -> Check for updates.

The other options:
- install fresh eclipse CDT via this [link](https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2023-12/R/eclipse-cpp-2023-12-R-win32-x86_64.zip) and then install our latest plugin via [update site](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.1).
- Download Espressif-IDE,  which already has all required dependencies installed.
Espressif IDE
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v2.12.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v2.12.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.1-win32.win32.x86_64.zip
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.1-linux.gtk.x86_64.tar.gz