espressif/idf-eclipse-plugin #876: Does not resolve any libraries

hcanak, 2023-12-21:
**Describe the bug**
**I want to develop an RFID application using esp32-c3. My rfid chip has its own library.
I created a blank project and copy the rfid libraries to the project folder. When i open the library files i see that it does not resolve include files.  How can i fix this issue? Thanks for all...

See picture please:**

**Espressif-IDE Product Information:**
Operating System: windows 10
Java Runtime Version: 21.0.1+12-LTS-29
Eclipse Version: 4.29.0.v20230903-1000
Eclipse CDT Version: 11.3.0.202307110141
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v4.4.6-dirty
Python set for IDF_PYTHON_ENV: Python 3.12.1

[error.log](https://github.com/espressif/idf-eclipse-plugin/files/13745987/error.log)

[image]

[image]

kolipakakondal, 2023-12-26:
Hi @hcanak Do you see this issue even after the build? Headers and symbols will be resolved only after the build.

hcanak, 2023-12-26:
> Hi @hcanak Do you see this issue even after the build? Headers and symbols will be resolved only after the build.

Yes. Building gives same errors.

kolipakakondal, 2023-12-26:
From the log, I see parser is failed.
> !ENTRY org.eclipse.cdt.core 4 0 2023-12-21 22:23:33.734
!MESSAGE Unexpected overload for binary operator 28: 'operator =='

Hi @hcanak If the build is sucessful and you still see headers and symbols issues, then this could be due to new C/C++ 20 language features used in the some of these files you have added, which is not yet supported by the Eclipse CDT Indexer.

I would suggest you can setup CDT LSP based editor which has a support for new language features. For setup, please refer to this [documentation](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md).

If the build not successful, then this might be an issue with configuring CMake build system. Please refer to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements).

hcanak, 2023-12-26:
I have uninstalled everthing and started a fresh install.
I gives the folloving error when i want to install LSP.

[image]

[image]

kolipakakondal, 2023-12-27:
Here is what you can do:
1. Download and install https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.0-win32.win32.x86_64.zip
2. Launch the IDE
3. Install LSP using this udpate site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
4. Restart IDE
5. Configure/Install ESP-IDF and Tools using menu "Espressif> ESP-IDF Tools Manager > Install Tools", here choose your existing esp-idf directory path, python and git and click on install.
6. Configure clangd as mentioned [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md)

Let us know how it goes.

hcanak, 2023-12-27:
> Here is what you can do:
> 
> 1. Download and install https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.0-win32.win32.x86_64.zip
> 2. Launch the IDE
> 3. Install CDT-LSP using this udpate site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
> 4. Restart IDE
> 5. Configure/Install ESP-IDF and Tools using menu "Espressif> ESP-IDF Tools Manager > Install Tools", here choose your existing esp-idf directory path, python and git and click on install.
> 6. Configure clangd as mentioned [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md)
> 
> Let us know how it goes.

All these steps success. 
I started example project but LSP does not seem to work, see below please:

[log.log](https://github.com/espressif/idf-eclipse-plugin/files/13780424/log.log)

[image]

kolipakakondal, 2023-12-28:
Hi @hcanak Syntax coloring seems to be missing, will take a look. 

But editor will detect syntax errors, provide content proposals and works well with the new C/C++ lang features. For navigation you can use F3 or Ctrl+Click.

hcanak, 2023-12-28:
> Hi @hcanak Syntax coloring seems to be missing, will take a look.
> 
> But editor will detect syntax errors, provide content proposals and works well with the new C/C++ lang features. For navigation you can use F3 or Ctrl+Click.

The project builds succesfully but F3 or CRTL+Click does not work;

[image]

hcanak, 2024-01-17:
Hello, 
I formatted windows and install everything several times, both in win10 and win11, the issue is same:(
Is there any suggestion? @kolipakakondal 

Is someone interested in this issue or should I consider to switch stm32?

sigmaaa, 2024-03-15:
Hi @hcanak,

Apologies for the delayed response. Fortunately, we were able to reproduce and address this issue by updating dependencies. You can find the details of the fix here: https://github.com/espressif/idf-eclipse-plugin/issues/892#issuecomment-1999786320.