espressif/idf-eclipse-plugin #880: OutputStreamMonitor.startMonitoring(java.lang.String)

Velvizhi123373, 2024-01-02:
OutputStreamMonitor.startMonitoring(java.lang.String) 
 During the Debug Configuration. We face this error it shows a pop Launching Configuration error 
[image]
Eclipse Version (2022-03)
Espressif esp_idf v4.4.6

**To Reproduce**
Steps to reproduce the behavior:
1. Right click your project Go to Debug.
2. Click on Debug Configuration
3.Start debug.
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

sigmaaa, 2024-01-02:
Hi @Velvizhi123373,

Thank you for reporting this. I haven't faced this issue before. Could you please provide an error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log? The Product information log (Espressif > Product Information (Copy content from the console and attach as a file)) will also help us investigate what could cause this issue.

Velvizhi123373, 2024-01-03:
[image]
[image]

I have attached the snip for error log.

Velvizhi123373, 2024-01-03:
[image]
Hi I have attached the product information log(The Product information log (Espressif > Product Information)

sigmaaa, 2024-01-03:
> ![image](https://private-user-images.githubusercontent.com/120127263/293874044-090f8e60-65ae-4ee4-bac2-307f6f62c4e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Hi @Velvizhi123373, 

Your product information log seems ok to me. We need the Exception Stack Trace from the error log to find the root of the problem. To do this, click on the "Export Log" icon here:
[image]

Velvizhi123373, 2024-01-04:
[image]
message : 
[image]
I have attached the  Exception Stack trace.

Velvizhi123373, 2024-01-04:
Message : C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/openocd.exe -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/share/openocd/scripts -c set ESP32_FLASH_VOLTAGE 1.8 -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32s2.cfg -c program_esp_bins C:/Users/C9950907/eclipse-workspace/blink/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"

sigmaaa, 2024-01-04:
> <img alt="Exception Stack Trace" width="950" src="https://private-user-images.githubusercontent.com/120127263/294098540-ec46ae7f-8d58-42f8-9ee9-a9ffb0d3fc49.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzNTg0ODksIm5iZiI6MTcwNDM1ODE4OSwicGF0aCI6Ii8xMjAxMjcyNjMvMjk0MDk4NTQwLWVjNDZhZTdmLThkNTgtNDJmOC05ZWU5LWE5ZmZiMGQzZmM0OS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNFQwODQ5NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZmM3YjI1ZDI4OWYyNGVlYmUxN2ZiMTE0ZTlkZTNlYjRmMmI0YzhkODdmZDAwOWUxMjk3OTUyN2NjNDY3ODhkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ATDLE8vW2bY0wNUu1prZH9HLMifCkEDHgfUcbXvuD9Q"> message : [image] I have attached the Exception Stack trace.

Hi @Velvizhi123373,

we need the Exception Stack Trace from the other error "An internal error occurred during: "Launching Blink Configuration", but it may be useful to have the whole error log with all errors as a text file attached here

sigmaaa, 2024-01-05:
> Hi I attached the image and also added the text which left in image please do refer.
> 
> <img alt="Internal log Error" width="943" src="https://private-user-images.githubusercontent.com/120127263/294223599-6d69b944-6551-46cc-af51-7a8992f05600.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0NDg4NjEsIm5iZiI6MTcwNDQ0ODU2MSwicGF0aCI6Ii8xMjAxMjcyNjMvMjk0MjIzNTk5LTZkNjliOTQ0LTY1NTEtNDZjYy1hZjUxLTdhODk5MmYwNTYwMC5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNVQwOTU2MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZWY4NTk2NGQ1ZmZlMThmNDdlZDA1OTk3NGQyYjhhZjM1MGIzMjIxNjdiNTY4MTI3NmExZGZkNDg1NmY1NThjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HM8dZgSV_s-s33eqtQkamvPHihmao7ppuzyxMQUmAsQ"> java.lang.NoSuchMethodError: 'void
> Exception Stack Trace Complete Details org.eclipse.debug.internal.core.OutputStreamMonitor.startMonitoring(java.lang.String)' at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.monitors.CustomOutputStreamMonitor.startMonitoring(CustomOutputStreamMonitor.java:32) at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.monitors.StreamsProxy.(StreamsProxy.java:76) at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.IdfRuntimeProcess.createStreamsProxy(IdfRuntimeProcess.java:62) at org.eclipse.debug.core.model.RuntimeProcess.(RuntimeProcess.java:150) at org.eclipse.cdt.dsf.gdb.launching.GDBProcess.(GDBProcess.java:35) at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.IdfRuntimeProcess.(IdfRuntimeProcess.java:35) at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.CustomIdfProcessFactory.newProcess(CustomIdfProcessFactory.java:26) at org.eclipse.debug.core.DebugPlugin.newProcess(DebugPlugin.java:816) at com.espressif.idf.debug.gdbjtag.openocd.dsf.Launch.addServerProcess(Launch.java:226) at com.espressif.idf.debug.gdbjtag.openocd.dsf.Launch.initializeServerConsole(Launch.java:174) at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launchDebugSession(LaunchConfigurationDelegate.java:472) at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launchDebugger(LaunchConfigurationDelegate.java:325) at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launch(LaunchConfigurationDelegate.java:296) at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:803) at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:716) at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1021) at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1224) at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
> 
> Session Data Complete Details:
> 
> eclipse.buildId=4.23.0.I20220308-0310 java.version=17.0.2 java.vendor=Eclipse Adoptium BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_IN Framework arguments: -product org.eclipse.epp.package.cpp.product Command-line arguments: -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

Hi @Velvizhi123373,

Thank you for providing the logs, now we understand the issue better. The issue happens due to incompatibility with the older Eclipse version. If you don't need to use the older version, I recommend you to install updates in Eclipse by using Help -> Check Updates, or if you need to be more specific with updates use Help -> Install New Software -> Work with -> CDT - https://download.eclipse.org/tools/cdt/releases/latest. The latest eclipse cdt and eclipse should be ok with the esp-idf plugin version that you are using (2.12)

The other option would be using Espressif-IDE, which has all the necessary plugins in it: https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.0-win32.win32.x86_64.zip 

If you need to use an older version of Eclipse (2022.03) which you are using now, you will have to downgrade the esp-idf plugin to 2.10 version https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.10.0

Velvizhi123373, 2024-01-08:
[image]
java.lang.NoSuchMethodError: 'void org.eclipse.debug.internal.core.OutputStreamMonitor.startMonitoring(java.lang.String)'
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.monitors.CustomOutputStreamMonitor.startMonitoring(CustomOutputStreamMonitor.java:32)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.monitors.StreamsProxy.<init>(StreamsProxy.java:76)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.IdfRuntimeProcess.createStreamsProxy(IdfRuntimeProcess.java:62)
	at org.eclipse.debug.core.model.RuntimeProcess.<init>(RuntimeProcess.java:150)
	at org.eclipse.cdt.dsf.gdb.launching.GDBProcess.<init>(GDBProcess.java:35)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.IdfRuntimeProcess.<init>(IdfRuntimeProcess.java:35)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.process.CustomIdfProcessFactory.newProcess(CustomIdfProcessFactory.java:26)
	at org.eclipse.debug.core.DebugPlugin.newProcess(DebugPlugin.java:816)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.Launch.addServerProcess(Launch.java:226)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.Launch.initializeServerConsole(Launch.java:174)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launchDebugSession(LaunchConfigurationDelegate.java:472)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launchDebugger(LaunchConfigurationDelegate.java:325)
	at com.espressif.idf.debug.gdbjtag.openocd.dsf.LaunchConfigurationDelegate.launch(LaunchConfigurationDelegate.java:296)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:803)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:716)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1021)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1224)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

After I installed the lastest version of Eclipse also I am facing the same issuse please do a check

sigmaaa, 2024-01-08:
Hi @Velvizhi123373,

Are you sure that the update was successful? I see from the error log, that eclipse.buildId still refers to the old version (20220308). Please, check the product information log, the Eclipse CDT should be also updated. After the update it should be like this:
``` Operating System: windows 11
Java Runtime Version: 17.0.2+8
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 2.12.0.202312220530
ESP-IDF v5.1.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.12.0
```
 What option to update did you use? 

Also, if eclipse is not updating for you for some reason, you can install a fresh installation of the latest Eclipse CDT: https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2023-12/R/eclipse-cpp-2023-12-R-win32-x86_64.zip

Also, did you consider to use Espressif-IDE, which has all the necessary plugins in it: https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.0-win32.win32.x86_64.zip?

Velvizhi123373, 2024-01-09:
[image]

Product Information :
Executing python C:\ESP-IDF\esp-idf-v4.4\tools\idf_tools.py list
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
ErrorError

CDT Build environment variables
PATH: C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\xtensa-clang\12.0.1-d9341b81fc\xtensa-esp32-elf-clang\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\python_env\idf4.4_py3.8_env\Scripts;C:\ESP-IDF\esp-idf-v4.4\tools;c:/users/c9950907/downloads/eclipse-cpp-2022-03-r-win32-x86_64/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.2.v20220201-1208/jre/bin/server;c:/users/c9950907/downloads/eclipse-cpp-2022-03-r-win32-x86_64/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.2.v20220201-1208/jre/bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\1E\Client\Extensibility\NomadBranch;C:\Program Files\Git\cmd;C:\Users\C9950907\AppData\Local\Microsoft\WindowsApps;C:\Users\C9950907\Downloads\eclipse-cpp-2022-03-R-win32-x86_64\eclipse
USERDOMAIN_ROAMINGPROFILE: EATON
PROCESSOR_LEVEL: 6
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE: 1
USERNAME: C9950907
USERDNSDOMAIN: EATON.AD.ETN.COM
ProgramFiles(x86): C:\Program Files (x86)
FPS_BROWSER_USER_PROFILE_STRING: Default
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
HOMEPATH: \Users\C9950907
PROCESSOR_IDENTIFIER: Intel64 Family 6 Model 78 Stepping 3, GenuineIntel
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\WINDOWS
=::: ::\
LOCALAPPDATA: C:\Users\C9950907\AppData\Local
USERDOMAIN: EATON
LOGONSERVER: \\SGIDCSPEATON22
FPS_BROWSER_APP_PROFILE_STRING: Internet Explorer
IDF_TOOLS_PATH: C:\Espressif
OneDrive: C:\Users\C9950907\OneDrive
APPDATA: C:\Users\C9950907\AppData\Roaming
CommonProgramFiles: C:\Program Files\Common Files
OS: Windows_NT
COMPUTERNAME: INCHNWHP6027219
IDF_PATH: C:\ESP-IDF\esp-idf-v4.4
PROCESSOR_REVISION: 4e03
CommonProgramW6432: C:\Program Files\Common Files
UATDATA: C:\WINDOWS\CCM\UATData\D9F8C395-CAB8-491d-B8AC-179A1FE1BE77
ComSpec: C:\WINDOWS\system32\cmd.exe
IDF_MAINTAINER: 1
TEMP: C:\Users\C9950907\AppData\Local\Temp
SystemRoot: C:\WINDOWS
HOMEDRIVE: C:
USERPROFILE: C:\Users\C9950907
TMP: C:\Users\C9950907\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 4
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf4.4_py3.8_env

Operating System: windows 10
Java Runtime Version: 17.0.2+8
Eclipse Version: 4.23.0.v20220308-0310
Eclipse CDT Version: 10.6.0.202201122357
IDF Eclipse Plugin Version: 2.12.0.202312220530
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7

Could You Please connect me in teams and resolve the issues my colleague did not face this issues.

Velvizhi123373, 2024-01-09:
[image]
While installing through the 1st link CDT-> latest/release  you provided.

Please connect me in teams and resolve the issue it would me more helpful.

AndriiFilippov, 2026-03-17:
resolved. @sigmaaa confirmed.