espressif/idf-eclipse-plugin #883: 'Install Tools' in 2.12.0 fails with ESP-IDF v5.2

gregtoth, 2024-01-09:
**Describe the bug**
'Install Tools' in 2.12.0 fails with ESP-IDF v5.2 beta releases.
When run Espressif > ESP-IDF Tools Manager > Install Tools, it always ends with Problem Occurred dialog containing the following message:
An internal error occurred during: "Installing tools...".
Cannot invoke "org.eclipse.cdt.core.build.IToolChain.getTypeId()" because "toolchain" is null

Also Preferences... > C/C++ > Core Build Toolchains is missing the Xtensa compiler entries.

Fails with esp-idf-v5.2-beta1 and esp-idf-v5.2-beta2.
Works okay with esp-idf-v5.1.2
Fails for both Espressif-IDF 2.12.0 and Eclipse 2023-12 with plugin 2.12.0 installed.

When using Espressif-IDF 2.11.1, Install Tools with esp-idf-v5.2-beta2 works okay.

**To Reproduce**
Steps to reproduce the behavior:
1. Delete any existing ~/.espressif directory.
2. Launch Espressif-IDE 2.12.0 or Eclipse 2023-12 with plugin 2.12.0 installed, and create a fresh new workspace.
3. Espressif > ESP-IDF Tools Manager > Install Tools
4. Select directory containing properly downloaded ESP-IDF v5.2-beta1 or v5.2-beta2.
5. Installation fails with Problem Occurred.
6. Inspect Preferences... > C/C++ > Core Build Toolchains and Xtensa entries are missing.

**Expected behavior**
Tools should install successfully and Xtensa toolchain settings should be populated.

When using Espressif-IDF 2.11.1, Install Tools with esp-idf-v5.2-beta2 works okay.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Espressif-IDF 2.12.0 or Eclipse 2023-12 with ESP-IDF Plugin 2.12.0 installed.

**Eclipse Error log:** 
[error-log-2.12.0-install-tools-2024-0109.log](https://github.com/espressif/idf-eclipse-plugin/files/13878309/error-log-2.12.0-install-tools-2024-0109.log)

sigmaaa, 2024-01-10:
Hi @gregtoth,

thank you for reporting this. We already have PR for this issue, which will be fixed as soon as possible. I will keep you updated. Thank you

BitSalat, 2024-01-18:
Same here on WIndows 11 - installation fails, I used latest installer **esp-idf-tools-setup-online-2.24.exe**

[image]

alirana01, 2024-01-23:
The fix is in the master should be out with the new release. Feel free to open the issue if it happens after the upcoming release @gregtoth @BitSalat

closed 2024-01-23 by alirana01

BitSalat, 2024-02-01:
Any idea when there will be the new release?

kolipakakondal, 2024-02-01:
Hi @BitSalat Next release will be most likely in the mid of Feb.