espressif/idf-eclipse-plugin #898: [Debian 12, Eclipse IDF 2023-12, ESP-IDF Plugin 2.12] ESP-IDF Tools manager > Install Tools errors.

GTCLive, 2024-02-06:
**Environment:** 
Linux Debian 12 (Bookworm, kernel: 6.1.0-17)
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2023-12 (4.30.0)
Build id: 20231201-2043
Espressif IDF Plugins for Eclipse	2.12.0.202312220530

**Steps to reproduce:**
Eclipse IDE (Top menu) > Espressif > ESP-IDF Tools manager > Install Tools

**ERROR LOG:** [2023-12-Install Tools Log.txt](https://github.com/espressif/idf-eclipse-plugin/files/14174683/2023-12-Install.Tools.Log.txt)

[image]

diplfranzhoepfinger, 2024-02-09:
same here!

this happens only with esp-idf master, not with esp-idf 5.1.2

diplfranzhoepfinger, 2024-02-09:
this seems to be an esp-idf Problem

GTCLive, 2024-02-16:
Fixes the qemu error and missing dependency:
`$ sudo apt install libslirp0`

GTCLive, 2024-02-16:
Fixes the qemu error and missing dependency:
$ sudo apt install libslirp0

closed 2024-02-16 by GTCLive

diplfranzhoepfinger, 2024-02-20:
```
sudo apt install libslirp0
```

diplfranzhoepfinger, 2024-02-20:
espressif/esp-idf#13152