espressif/idf-eclipse-plugin #899: install error

peterkosek, 2024-02-06:
I have installed the Espressif_eclipse IDE, and the tools are not available from within eclipse.  

I used the windows installer, and the CMD and Powershell command lines seem to work idf.py commands like build and flash.  

On entering eclipse, following the instructions at 'https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/Espressif-IDE-Windows-Installer.md', I found no environment variables set, unlike the guide.  I was offered to install the tools from within eclipse.    

I attempted to install the tools from the Espressif menu in Eclipse, and  was given

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 2687, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 2679, in main
    action_func(args)
  File "C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 1625, in action_export
    tools_info = load_tools_info()
                 ^^^^^^^^^^^^^^^^^
  File "C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 1296, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:  # type: ignore
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Espressif_eclipse\\esp-idf-v5.1\\tools/tools.json'

Here is the issue, this last directory is an old install that was removed.  I can not find any reference to this directory in my environment variables.  Why is it looking there?

I have used eclipse in the past with other frameworks, but deleted those from eclipse, perhaps a hidden PATH variable from those installs is causing this problem.

I appreciate any help.

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

alirana01, 2024-02-06:
Hi @peterkosek thanks for reporting the issue can you please try to run this bat script for command line 

```
@echo off
setlocal EnableDelayedExpansion

REM Loop through all environment variables
for /f "tokens=1* delims==" %%a in ('set') do (
    REM Check if the variable name contains "idf", ignoring case
    echo %%a | findstr /I "idf" >nul
    if !errorlevel! == 0 (
        REM Print the variable name and its value
        echo %%a=%%b
    )
)
endlocal
```

To use this script:

1. Open Notepad or any text editor.
2. Copy and paste the script into the editor.
3. Save the file with a .bat extension, for example, PrintIDFEnvVars.bat.
4. Run the script from the command prompt.

Please share the output

peterkosek, 2024-02-06:
C:\Users\pkose\Documents>esp32Script.bat
IDF_PATH=C:\Espressif\frameworks\esp-idf-v5.1.2
IDF_PYTHON_ENV_PATH=C:\Espressif\frameworks\tools\Espressif\python_env\idf5.1_py3.11_env
IDF_TOOLS_PATH=C:\Espressif

C:\Users\pkose\Documents>

Thank you

alirana01, 2024-02-06:
@peterkosek so you need to remove these env variables to be specific I believe removing only IDF_TOOLS_PATH from your env variables in windows should be enough. Please restart the IDE if its already opened after removing these variables and if you are starting it from command line also restart that command line once you have removed these variables from windows environment

peterkosek, 2024-02-07:
Thank you for your help.

I deleted all of the system user variables that started with 'IDF', and then the tools installed and I am up and compiling.  

Thanks for your prompt help.

alirana01, 2024-02-07:
@peterkosek you are very welcome glad to be of help.
Please feel free to reopen the issue if you have any trouble again with this.

closed 2024-02-07 by alirana01