espressif/idf-eclipse-plugin #902: Eclipse Plugin: Add ability to select ESP-Tools location

christiankarsch, 2024-02-19:
*Windows 10 64bit
*Eclipse CDT 2023-12
*ESP-IDF plugins for Eclipse 2.3.0
*ESP-IDF 5.1.2

An existing ESP-IDF location can be specified, that's good and it also works.

But the additional ESP tools must be installed via the Eclipse plugin (ESP-IDF Tools Manager), there is no option to specify the installation location.

The location will always be %USERPROFILE%\.espressif

Even if the tools have already been installed manually.

On our company computers, all applications that are not located in certain locations are blocked from running. E.g. "C:\Program Files" is allowed
The user directory and the temp directories are not included.

Even without admin rights, we have write rights to a specific folder to manually install programs there.
I also installed the ESP-IDF and the ESP tools there. But I cannot specify the existing ESP-Tools folder in Eclipse.

kolipakakondal, 2024-02-27:
Hi @christiankarsch The recommended approach for now would be to configure the IDF_TOOLS_PATH environment variable at the system level/user level. Afterward, restart Eclipse and proceed to install the tools. They will be installed in the custom-configured tools path.

In future releases, we will provide an option to configure Tools installation path in the install tools dialog itself to have a better usablity.

kolipakakondal, 2024-03-22:
@alirana01 Please include this in esp-idf managers view implementation.

We could provide two options:

Workspace-level configuration for all ESP-IDF installations (managed through preferences).
ESP-IDF configuration level, where users could select a specific tools location for each ESP-IDF (managed through wizard configuration).

At least we can target option 1 for IEP v2.13.0.

AndriiFilippov, 2026-04-17:
EIM now covers this functionality.