espressif/idf-eclipse-plugin #907: Add the Reset & Restart functionality

LemonBoy, 2024-03-01:
Some other IDEs by other uC vendors include a "Reset & Restart" button allowing the devloper to restart both the uC and the debugging session without having to reflash the image every time.

Under the hood the command is simply sending a `monitor reset` command to the GDB instance the debugger is attached to, AFAICT the OpenOCD/GDB interface is already working but issuing that command from the "Debugger Console" confuses the debugger.

sigmaaa, 2024-03-05:
Hi @LemonBoy,

Thank you for reporting this and sorry for delay response. Have you tried this button? It restarts debug without terminating and relaunching. 
[image]

It's also available from the right click on the debug thread action:
[image]

LemonBoy, 2024-03-05:
That's exactly the button I was referring to, but on my Eclipse (or better, Espressif IDE) install there's no trace of such button:
[image]

I can only guess it's because the ESP plugin and Eclipse are slightly old, I tried re-installing it using the latest online Installer but got the same 2.9.1 version again.

```
Espressif-IDE is an Integrated Development Environment(IDE) based on Eclipse CDT for developing IoT Applications using ESP-IDF framework.
Version: 2.9.1
Build id: 20230406-1540
OS: Windows 11, v.10.0, x86_64 / win32
Java vendor: Amazon.com Inc.
Java runtime version: 17.0.6+10-LTS
Java version: 17.0.6
```

LemonBoy, 2024-03-05:
Well, it turns out that Eclipse removed that button from the toolbar a while ago. No idea why it's still present in your Eclipse install.

From the plugin side it should be quite easy to add back the button into the debug toolbar, the action name is `org.eclipse.debug.ui.commands.Restart`.

sigmaaa, 2024-03-05:
Hi @LemonBoy,

Thank you for the info. I didn't know this command wasn't enabled by default. I probably have it because I accidentally installed additional (optional) packages from CDT Eclipse via the update site. We will include this command by default.

By the way, thank you for pointing out that the Espressif-IDE version is not updated in the online installer. It looks like PR is already there and will be merged soon. Here are options that you can use now to get the latest Espressif-IDE version:
- Espressif IDE 2.12.1 and then go to Espressif -> ESP-IDF Tools Manager -> Install Tools 
- Windows offline installer - https://github.com/espressif/idf-installer/releases/download/espressif-ide-2.12.1-esp-idf-5.2/espressif-ide-setup-2.12.1-with-esp-idf-5.2.exe
- install the latest Eclipse CDT and then our plugin via update site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

sigmaaa, 2024-03-05:
Hi @LemonBoy,

I found which plugin update has `org.eclipse.debug.ui.commands.Restart action`. It's `Embedded C/C++ GDB JTAG Debugging` available in this update site: https://download.eclipse.org/embed-cdt/updates/v6/
or by using https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ as in the screenshot below:
[image]

I tested it by installing the latest version of Espressif-IDE and after updating this plugin, this action appeared in the toolbar

LemonBoy, 2024-03-07:
Thanks for the feedback, after installing the updated `Embedded C/C++ GDB JTAG Debugging` package the restart action is shown in the toolbar.

Shall I close the issue? Or do you still plan on doing something related to this?

sigmaaa, 2024-03-07:
Hi @LemonBoy,

We are planning to make this action available out of the box. Until then, let's keep this issue open. Thank you.

FabITA1, 2024-11-08:
I have the same issue. 
Sorry, but I don't understand how to fix.

FabITA1, 2024-11-15:
@GrantGeno thanks for your supporto. It works also for me. Good!