espressif/idf-eclipse-plugin #913: Latest version brings more unresolved warnings

dizcza, 2024-03-12:
**Describe the bug**
After I've updated the plugin from v2.2 to v2.12, I see more errors (unresolved references), cannot navigate through the code. Even things like `uint8_t` and `esp_err_t` cannot be resolved. Working in such an environment has become hard.

The project compiles without a problem though.

**To Reproduce**
Steps to reproduce the behavior:
1. Update the plugin.
2. Open a new ESP-IDF project.
3. Build (optionally).

**Expected behavior**
No unresolved errors highlighted.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
```
Executing /home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python3 /home/dizcza/tools/esp-idf/tools/idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20231023 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20231023 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20230921 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
* qemu-xtensa: QEMU for Xtensa (optional)
  - esp_develop_8.1.3_20231206 (recommended, installed)
* qemu-riscv32: QEMU for RISC-V (optional)
  - esp_develop_8.1.3_20231206 (recommended, installed)
OK

CDT Build environment variables
PATH: /home/dizcza/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/dizcza/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/dizcza/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/dizcza/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/dizcza/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/dizcza/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin:/home/dizcza/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/dizcza/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/dizcza/.espressif/tools/cmake/3.24.0/bin:/home/dizcza/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/dizcza/.espressif/tools/ninja/1.10.2/:/home/dizcza/.espressif/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin:/home/dizcza/.espressif/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin:/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin:/home/dizcza/tools/esp-idf/tools:/home/dizcza/tools/google-cloud-sdk/bin:/home/dizcza/miniconda3/bin:/home/dizcza/miniconda3/condabin:/home/dizcza/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/dizcza/tools/winetricks/src:/media/dizcza/media/Android/Sdk/platform-tools:/usr/local/cuda/bin:/usr/local/go/bin:/home/dizcza/go/bin:/home/dizcza/.local/bin
XMODIFIERS: @im=ibus
GDMSESSION: ubuntu
CONDA_DEFAULT_ENV: esp
CONDA_PYTHON_EXE: /home/dizcza/miniconda3/bin/python
ENV_OTA_SERVER_USER: spivoche
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
ESP_IDF_VERSION: 5.1
DEFAULTS_PATH: /usr/share/gconf/ubuntu.default.path
ANDROID_HOME: /home/dizcza/Android/Sdk
CONDA_PREFIX: /home/dizcza/miniconda3/envs/esp
IDF_DEACTIVATE_FILE_PATH: /tmp/tmpr0w3dsaxidf_19110
COLORTERM: truecolor
STY: 27593.eclipse
USERNAME: dizcza
SESSION_MANAGER: local/ubu:@/tmp/.ICE-unix/2053,unix/ubu:/tmp/.ICE-unix/2053
LC_PAPER: uk_UA.UTF-8
PWD: /home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor
[... 82 lines cut ...]
TF_XLA_FLAGS: --tf_xla_cpu_global_jit
GPG_AGENT_INFO: /run/user/1000/gnupg/S.gpg-agent:0:1
USER: dizcza
GIO_LAUNCHED_DESKTOP_FILE_PID: 13377
WINDOWPATH: 2
LC_NUMERIC: uk_UA.UTF-8
TERMINATOR_UUID: urn:uuid:ec41aeda-735c-4a4c-bdb1-f10eae7f592d
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
_CE_CONDA: <NOT FOUND>
IDF_MAINTAINER: 1
GNOME_SHELL_SESSION_MODE: ubuntu
XDG_RUNTIME_DIR: /run/user/1000
HOME: /home/dizcza

Operating System: linux
Java Runtime Version: 17.0.3+7
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 2.12.1.202402231119
ESP-IDF v5.1.3-204-ge3e112e612
Python set for IDF_PYTHON_ENV: Python 3.9.12
```

**Eclipse Error log:** 
[errors.log](https://github.com/espressif/idf-eclipse-plugin/files/14578625/errors.log)

dizcza, 2024-03-12:
Previously, I had such problems mostly in C++ files (ESP_LOGX were always highlighted, includes, `static const char *TAG` has always been rendered as unused, etc.) but now I see these warnings even in C files.

dizcza, 2024-03-12:
Tried installing ESP-IDF again after the plugin was updated - didn't help.

kolipakakondal, 2024-03-13:
Hi @dizcza Could you share a sample application which I download and try?

dizcza, 2024-03-13:
Oh, when I deleted the project and imported it again, everything was resolved. Perhaps, I ignored the warning after Eclipse update about the old workspace directory and clicked continue.

Now what is being highlighted are atomics (have always been, I remember you said it's not on your side) and `esp_log`s (only for C++ files). C++ files are resolved much better now compared to v2.2, where I had plenty of highlights.

You can test the esp_log yourself:
```cpp
#include "esp_log.h"

extern "C" { void app_main(); }

void app_main(void)
{
    ESP_LOGI(TAG, "main init");
    while (1) {
        ;
    }
}
```

Both `TAG` and `ESP_LOGI` are highlighted as unused and a syntax error respectively.

----

While I was happy writing it, I still hit the same issue with a completely new project. Steps to reproduce:
1. Create a new ESP-IDF project from samples and choose `sample_project`, rename it to `test_espidf_plugin`.
2. Move `main.c` to `main.cpp` and update the cmake accordingly.
3. Write the code to `main.cpp`.
4. Build. (At this point, everything was resolved correctly - no highlighting errors save for `esp_log`s)
5. Import another ESP-IDF project, select it from the drop-down build configuration, click build.
6. Now the imported project is resolved OK.
7. Switch back to the `test_espidf_plugin` project and you'll see highlighted errors again. Clean and rebuild didn't help.

[image]

I needed to re-import `test_espidf_plugin` again to resolve the problem though it was created a minute ago.

P.S. Funny, among a huge list of syntax errors in the screenshot above, the `TAG` is not highlighted :)

sigmaaa, 2024-03-13:
Hi @dizcza,

> Switch back to the test_espidf_plugin project and you'll see highlighted errors again. Clean and rebuild didn't help.

Have you tried closing/reopening the project as a workaround? Usually switching projects in the launch bar causes the indexer to rerun, after the indexer's job is finished, it should be fine, but I'm facing the same issue sometimes, not sure why it happens and what confuses the indexer, but closing and reopening project helps me in such scenarios.

dizcza, 2024-03-13:
> Have you tried closing/reopening or closing/reopening/building the project as a workaround?

I tried both and none of them helped. Only deleting the project and importing it again did the job.

closed 2024-03-13 by dizcza

sigmaaa, 2024-03-13:
Hi @dizcza,

I forgot to mention, that we are working on LSP integration right now, since Eclipse CDT Indexer is not actively maintained, and doesn't support C/C++ 20 (which is used in ESP-IDF 5.0) I think Clangd indexer should work better. It's easy to set by following this guide: https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md

dizcza, 2024-03-13:
Well, I don't have the `Editor (LSP)` option in `C/C++` preferences once I installed the Eclipse CDT-LSP...

dizcza, 2024-03-14:
Ah, it wasn't installed properly. After I installed it again (somehow, in the previous attempt, the "accept source" window didn't even pop up), I set the paths as suggested. However, I often change the esp-idf version - I pull source code from the master branch, sometimes need to reinstall esp-idf, which creates a new python env and hence I need to repeat the same steps in the eclipse plugin. Previously, I needed to provide only one path - the python path. But now I need to provide three paths:
1. Path to python
2. Path to clang
3. Path to xtensa-gcc

That's quite a headache for me.

Also, after I followed the instructions and set the paths correctly, deleted the project, imported it again, built, but again I'm hitting the same problem I outlined in the first comment: too many highlights, indexer cannot resolve almost anything. Tried re-index, didn't help.

sigmaaa, 2024-03-14:
> However, I often change the esp-idf version 

We are working on new tools manager that will allow you to switch between esp-idf versions quicker https://github.com/espressif/idf-eclipse-plugin/pull/908

> Previously, I needed to provide only one path - the python path. But now I need to provide three paths

If you have multiple projects and each of them requires different esp-idf version, for now you can create different workspaces for different esp-idf versions, or you can configure LSP settings on project level, so you don't have to change path to clang and path to xtensa-gcc each time you are changing your esp-idf (right click on project ->Properties -> Editor LST -> "Enable project-specific settings".  We will make this process smoother with futher integration of CDT LSP plugin

> I'm hitting the same problem I outlined in the first comment: too many highlights, indexer cannot resolve almost anything

Do you have the same problem even with the clangd enabled?

dizcza, 2024-03-14:
> If you have multiple projects and each of them requires different esp-idf version, for now you can create different workspaces for different esp-idf versions, or you can configure LSP settings on [...]

I set only one esp-idf version for all of my projects, it's just I update it often. And if I afterwards don't re-install esp-idf inside the eclipse plugin, then code navigation leads to a wrong line.

> Do you have the same problem even with the clangd enabled?

Yes. For instance, "Could not find symbol 'esp_err_t' in index"

dizcza, 2024-03-14:
Sorry, it seems the LST wasn't properly enabled. I remember hitting Apply and close once I provided LST paths but next time when I opened the LST preferences, the "Prefer C/C++ LST" checkbox wasn't ticked and when I enabled it, the paths were there - how is this even possible?...

Anyhow, I reimported the project, and now all C files are displayed as text. And C++ files still have indexer issues

[image]
[image]

dizcza, 2024-03-14:
The default CDT indexer is off.

sigmaaa, 2024-03-15:
Hi @dizcza,

The first screenshot appears to show highlights from the regular indexer (LSP colors are different). Perhaps this issue will be resolved by addressing the case related to the second screenshot. From the error log you provided, it seems that the absence of the tm4e dependency is causing the second screenshot (no syntax highlights). Could you please try to obtain this dependency from the following update site: https://download.eclipse.org/tm4e/releases/latest/? To do so, navigate to Help -> Install New Software -> Work with (provide update site here), and then select Batik CSS engine.
[image]

Also, in the error logs, there are numerous methods that couldn't be loaded for some reason. For example:
'com.espressif.idf.ui.handlers.DeleteResourceHandler' could not be loaded. At present, I'm uncertain about the cause of this. I will update you on this later.

If updating the tm4e dependency doesn't resolve this issue, would you be open to connecting so we can address these issues together? Please feel free to contact us via email at ide.support@espressif.com. In your email, mention the time that suits you best, and we will send you an invitation to the meeting on Teams.

dizcza, 2024-03-20:
I switched back to the default indexer for now, it works fine.
I'll consider using your custom indexer if I experience annoying problems again.

dizcza, 2024-03-21:
I'm switching again to your custom editor - the default one breaks often.
After I switch to LSP, how to change the theme? Clicking ThemeMate doesn't do anything. I cannot uncheck the default checkbox for a particular theme and save it - it does not get saved.

dizcza, 2024-03-21:
CTRL+/ does not comment/uncomment the line - expected?

sigmaaa, 2024-03-21:
Hi @dizcza,

> CTRL+/ does not comment/uncomment the line - expected?

No, thank you for reporting this. Looks like some conflict with TM4 Language Configuration and C/C++ editor commands. To fix, please open Preferences -> General -> keys -> in filter type "tm4" -> select "Toggle Line Command" -> Click Copy and then set binding to Ctrl + / :
[image]

> After I switch to LSP, how to change the theme? Clicking ThemeMate doesn't do anything. I cannot uncheck the default checkbox for a particular theme and save it - it does not get saved.

Do you mean some custom theme? I haven't checked custom themes yet, but the default ones that are available in this list work fine for me:
[image]

There is no need to uncheck the "Default" checkbox, I'm just selecting a new theme, then on the checkbox "Default" to make this theme the new default, and after applying my editor colors are updated immediately

dizcza, 2024-03-22:
> There is no need to uncheck the "Default" checkbox, I'm just selecting a new theme, then on the checkbox "Default" to make this theme the new default, and after applying my editor colors are updated immediately

So that's the thing, "Apply" does not change the theme I'm selecting in TextMate themes

dizcza, 2024-03-22:
So I have both  Solarized Light and Dark themes marked as Default. They cancel each other, and perhaps Solarized Light wins but I'd like to have Dark. Dark is already default, just as Solarized Light. Unchecking the default checkbox in Solarized Light doesn't anything, applying doesn't do anything.

sigmaaa, 2024-03-22:
> So I have both Solarized Light and Dark themes marked as Default. They cancel each other, and perhaps Solarized Light wins but I'd like to have Dark. Dark is already default, just as Solarized Light. Unchecking the default checkbox in Solarized Light doesn't anything, applying doesn't do anything.

To switch the theme from light to dark you need to select the "Dark" theme in General -> Appearance -> Theme combo. So for a dark appearance only dark TextMate configs work and vice versa

dizcza, 2024-03-22:
> To switch the theme from light to dark you need to select the "Dark" theme in General -> Appearance -> Theme combo. So for a dark appearance only dark TextMate configs work and vice versa

Ah, so that's how it works.

All right. So far so good.

A couple of minor issues:

## 1

In a header file

`#include "freertos/FreeRTOS.h"` gets highlighted with the message "__GLIBC_USE not defined (included in /usr/include/limits.h 145:5)"

This happens when the corresponding source file for this header is C++, not C. C files (sources and headers) don't show this error. Wrapping this include line with `extern "C"` does not help.

When I click and navigate to "FreeRTOS.h", in the first line (starting with a doc string) I get
* Unknown argument -fno-tree-switch-conversion
* Unknown argument -fstrict-volatile-bitfields

My `.clangd`:

```
CompileFlags:
  Remove: [-fno-tree-switch-conversion, -fstrict-volatile-bitfields]
```

## 2 int types

```
static int32_t m_task_global_id = 0;
ESP_LOGI(TAG, "loaded task global id: %ld", m_task_global_id);
```

"%ld" seems the correct way to go for a build: if I change it to "%d", the build fails with the message 

> error: format '%d' expects argument of type 'int', but argument 6 has type 'int32_t' {aka 'long int'} [-Werror=format=]
 1717 |     ESP_LOGI(TAG, "loaded task global id: %d", m_task_global_id);

But your indexer keeps it highlighted with "%ld":

> Format specifies type long but the argument has type int32_t aka int

Same for `%lu` -> `%u`, etc. A bit annoying. The default indexer doesn't behave like that.

dizcza, 2024-04-03:
Well, I must say I have experienced pleasure of coding ever since I switched to your custom CDT indexer. It was hard to set up but with your help, it seems to be functioning properly. I like function argument name expansion, suggestions as you type, better C++ support (I don't need to delete the project and import it again) but most of all I no longer need to delete an extra prefix space that was added previously in situations like

```c
#ifdef __cplusplus
 extern "C" {
#endif

 // note the space before the function
 void my_func();

#ifdef __cplusplus
}
#endif
```

such a negligible thing but I feel relief not fixing each line manually now...

Just fix those minor issues and go with it.

I hope you guys make it stable and it'll become the standard.