espressif/idf-eclipse-plugin #919: IDE installation error

ertekinsener, 2024-03-21:
I followed the steps in this video, but IDE could not correctly installed.

Issue video: https://www.youtube.com/watch?v=mp-FvgQa78E

The IDE had to create a new project, but it give errors

[product_information.txt](https://github.com/espressif/idf-eclipse-plugin/files/14696527/product_information.txt)

 [error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/14696562/error_log.log)

sigmaaa, 2024-03-21:
Hi @ertekinsener,

Thank you for reporting this issue. I've noticed from the error log that tools.json location is incorrect "C:\\Espressif\\frameworks\\esp-idf-v5.1.1\tools" when you are using different esp-idf. Maybe there is some conflict in the PATH. 
Could you please try this:
1. In the IDE go to Preferences -> C/C++ -> Build -> Environment and click "Restore Defaults" button:
[image]
2. After this, once again try to install tools via Espressif -> ESP-IDF Tools Manager -> Install Tools. (Since tools are already installed, the installation will be skipped and only environment variables inside the IDE will be updated.

kolipakakondal, 2024-03-21:
Hello,

Look at the log, your system is configured with IDF_PATH=C:\Espressif\frameworks\esp-idf-v5.1.1
Could you please remove this variable and restart the eclipse, and see if that helps? 

We will work on the fix for this.

ertekinsener, 2024-03-23:
Hello @sigmaaa and @kolipakakondal,

I attempted to resolve the issue by removing IDF_PATH=C:\Espressif\frameworks\esp-idf-v5.1.1 from the environment variables and restarting my PC. However, this did not resolve the error. Subsequently, I uninstalled both Espressif IDE and IDF components that came with the IDE. Afterward, I reinstalled Espressif IDE.

Upon attempting to create a new IDF project, I encountered an error prompting me to install tools. I proceeded to reinstall the tools. Notably, none of the tools were skipped during the reinstallation process, and they were all installed in different directory. @sigmaaa 

I will attach the error logs once again. Thank you for your assistance.

IDF Tools Console:
Installing tools...
This can take a while. Please be patient.
Executing C:\Espressif_IDE\tools\idf-python\3.11.2\python.exe C:\Espressif_IDE\frameworks\esp-idf-v5.1.2\tools\idf_tools.py install all
Current system platform: win64
Updating C:\Users\Ertekin\.espressif\idf-env.json
Selected targets are: esp32, esp32s3, esp32c3, esp32h2, esp32c2, esp32c6, esp32s2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Installing xtensa-esp-elf-gdb@12.1_20221002
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\xtensa-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\xtensa-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002
Installing riscv32-esp-elf-gdb@12.1_20221002
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/riscv32-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\riscv32-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\riscv32-esp-elf-gdb-12.1_20221002-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002
Installing xtensa-esp32-elf@esp-12.2.0_20230208
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/xtensa-esp32-elf-12.2.0_20230208-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\xtensa-esp32-elf-12.2.0_20230208-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\xtensa-esp32-elf-12.2.0_20230208-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208
Installing xtensa-esp32s2-elf@esp-12.2.0_20230208
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/xtensa-esp32s2-elf-12.2.0_20230208-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\xtensa-esp32s2-elf-12.2.0_20230208-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\xtensa-esp32s2-elf-12.2.0_20230208-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208
Installing xtensa-esp32s3-elf@esp-12.2.0_20230208
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/xtensa-esp32s3-elf-12.2.0_20230208-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\xtensa-esp32s3-elf-12.2.0_20230208-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\xtensa-esp32s3-elf-12.2.0_20230208-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208
Installing esp-clang@15.0.0-23786128ae
Downloading https://github.com/espressif/llvm-project/releases/download/esp-15.0.0-20221201/llvm-esp-15.0.0-20221201-win64.tar.xz
Destination: C:\Users\Ertekin\.espressif\dist\llvm-esp-15.0.0-20221201-win64.tar.xz.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\llvm-esp-15.0.0-20221201-win64.tar.xz to C:\Users\Ertekin\.espressif\tools\esp-clang\15.0.0-23786128ae
Installing riscv32-esp-elf@esp-12.2.0_20230208
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-x86_64-w64-mingw32.zip
Destination: C:\Users\Ertekin\.espressif\dist\riscv32-esp-elf-12.2.0_20230208-x86_64-w64-mingw32.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\riscv32-esp-elf-12.2.0_20230208-x86_64-w64-mingw32.zip to C:\Users\Ertekin\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208
Installing esp32ulp-elf@2.35_20220830
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp32ulp-elf-v2.35_20220830/esp32ulp-elf-2.35_20220830-win64.zip
Destination: C:\Users\Ertekin\.espressif\dist\esp32ulp-elf-2.35_20220830-win64.zip.tmp

[==================================================] 100%Done
Extracting C:\Users\Ertekin\.espressif\dist\esp32ulp-elf-2.35_20220830-win64.zip to C:\Users\Ertekin\.espressif\tools\esp32ulp-elf\2.35_20220830
[... cut ...]
Collecting websocket-client
  Downloading websocket_client-1.7.0-py3-none-any.whl.metadata (7.9 kB)
Downloading websocket_client-1.7.0-py3-none-any.whl (58 kB)
   ---------------------------------------- 58.5/58.5 kB 518.5 kB/s eta 0:00:00
Installing collected packages: websocket-client
Successfully installed websocket-client-1.7.0
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

[error_logs.log](https://github.com/espressif/idf-eclipse-plugin/files/14733375/error_logs.log)