espressif/idf-eclipse-plugin #956: Move 'ESP-IDF: ****' context menu items into 'ESP-IDF' SUBMENU instead of putting them on top level

powerbroker, 2024-05-19:
[image]

When context menu cannot completely reside within screen area, it just sucks.

From another side there is absolutely no reason to put all of them on top level. The ''ESP-IDF:' prefix implies: they all must be UNDER the "ESP-IDF" submenu.

kolipakakondal, 2024-05-21:
Hi @powerbroker Agree with you. We are working on this, it will be part of the next release.

kolipakakondal, 2024-05-30:
This is fixed in the (https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.0.0-beta2 release, where all the menu items moved to ESP-IDF main menu.

[image]

kolipakakondal, 2024-06-06:
This is fixed in the v3.0.0 release, hence closing the issue.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.0.0

closed 2024-06-06 by kolipakakondal

powerbroker, 2024-06-15:
verified, OK