espressif/idf-eclipse-plugin #989: Espressif IDE error in help autocomplete

Ivan-Vargas-Alvarado, 2024-06-14:
**Error description**
I have Espressif IDE development tools installed since, espressif-ide-setup-3.0.0-with-esp-idf-5.2.2.
When I open Espressif IDE and create a project everything works fine with autocompletion (either when I try to include a header file or when I declare a structure type variable). Once you build the project for the first time, then the autocomplete stops working and never works again until you create a new esp-idf project. In the message in the attached image I get this error java.util.concurrent.CancellationException I have repeated the reinstallation many times and the problem is the same.

**Expected behavior**
I need autocomplete to work because it is an important help when programming. I need to know if there are any manual settings that have to be entered for it to work well.

**Screenshots**

Try using autocomplete
[image]

After a while the error occurs
[image]

**Espressif-IDE Product Information:**
Operating System: windows 10
Java Runtime Version: 17.0.6+10-LTS
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.2.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.2

danielealki, 2024-09-20:
Did someone saved this problem ?

sigmaaa, 2024-09-20:
Hi @danielealki,

I don't see this issue on the master version. You can try it by using the artifacts down below on this page: https://github.com/espressif/idf-eclipse-plugin/actions/runs/10849628637. Let me know if you experience this issue there as well. Thanks