espressif/idf-eclipse-plugin #991: Disable inlay hints (in .clangd project config) by default

powerbroker, 2024-06-15:
LSP Editor, used for C/C++ files of ESP-IDF project shows inlay hints, e.g. parameter names in function invocations.

looks and behaves suxy: they appears with a delay, they reside within editor area but NOT EDITABLE and they VIOLATE language syntax.

moreover, there is no option to prohibit such annoying staff in Eclipse options. the only way to get rid of them is '.clang' file in project root or 'config.yaml' in user home folder - https://clangd.llvm.org/config#files.
since affecting global level through user home is bad idea, the 

InlayHints:
  Enabled: No

inside project's '.clangd' file is a lifesaver.

**To Reproduce**
just open C/C++ file with some function call in LSP Editor of Eclipse.
enjoy appearing of parameter names before parameters in function invocations and in places like them after some delay.

**Expected behavior**
all-the-text within editor area is fully editable. no syntax-breaking (even visually) inline constructs appear: what-you-see-is-what-you(=compiler)-get.
everyone who needs this feature may ENABLE it, e.g. somewhere in Eclipse preferences.

**Screenshots**
[image]

note separate area for line numbers on the left and color markers on the right for different kinds of hints

**Espressif-IDE Product Information:**
Operating System: windows 11
Java Runtime Version: 21.0.3+9-LTS
Eclipse Version: 4.32.0.v20240601-0610
Eclipse CDT Version: 11.6.0.202403071917
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.2.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.9.16

IJeffray, 2024-08-09:
YES!  This has been infuriating me since it was introduced.  It's an awful feature and I'm boggled there's no UI preference to disable this - so THANK YOU for explaining how to at least hack the .clangd file and get rid of this.

mmrein, 2024-11-21:
+1 for disable by default. 

Also thanks for the hint on how to disable, uh, hints.