espressif/idf-eclipse-plugin #992: Bullshitty error message about format specifier

powerbroker, 2024-06-15:
**Describe the bug**
specifying printf("%lu", ...)" for value of 'uint32_t' results in Eclipse error message 'Format specifies type unsigned long but argument is of type unsigned int...' even if code WAS BUILT SUCCESSFULLY.
applying suggested fix - change '%lu' to '%u' - removes error message but BUILD FAILS with format error.

**To Reproduce**
just compile printf from screenshot
enjoy fake error / build failure of 'correct' code

**Expected behavior**
successfully compiled code HAS NO ERRORS like this.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Operating System: windows 11
Java Runtime Version: 21.0.3+9-LTS
Eclipse Version: 4.32.0.v20240601-0610
Eclipse CDT Version: 11.6.0.202403071917
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.2.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.9.16

gtjoseph, 2024-08-10:
The whole switch to the LSP editor is a complete disaster but there is a workaround for this one.  Use the `PRIu32` macro instead of `%u` or `%lu` ...

`sprintf((char *)UART_IO.dataTx, "%" PRIu32 ";" PRIu32 "",`

You might need to include `<inttypes.h>` to get the macros.

juantxorena, 2024-09-22:
I've being hit by this very stupid bug now. Probably some misconfiguration on the settings of eclipse, but I have no idea how to fix it.