espressif/idf-eclipse-plugin #994: Espressi-IDE 3.0.0 doesn't highlight inactive code

dariomingarelli, 2024-06-17:
Espressi-IDE 3.0.0 doesn't highlight inactive code.
I use a lot of #define and #ifdef, till 2.12.1 the inactive code were highlighted (for example, in black background for dark mode), now there is no way to have this working.

I've also tried to restore or modify the highlight color in:
Espressi-IDE > Preferences > C/C++ > Editor
And manage "Inactive code highlight" under "Appearance color options:".

luciantodor, 2024-06-18:
I subscribe to this issue. The inactive code is not highlighted at all.

closed 2024-10-14 by kolipakakondal

ifedotov1984, 2024-10-23:
> This is fixed with the v3.1.0 release which comes with the CDT LSP 2.0 which has the support for inactive code is highlight

Still do not work in Dark Theme (in Light all is ok)
[image]

kolipakakondal, 2024-10-23:
Ok, thanks for verifying and letting us know. Let us check.