espressif/idf-eclipse-plugin #995: Espressi-IDE 3.0.0 doesn't have "Open declaration" in context menu

dariomingarelli, 2024-06-18:
Espressi-IDE 3.0.0 doesn't have "Open declaration" in context menu.
I try also "F3" but doesn't work.

To reproduce:
A) highlight any var and click with mouse "right button" (two finger click on touchpad for MacOs) and search for "Open declaration".
B) highlight any var and press "F3"

zjanosy, 2024-06-18:
Use Ctrl-click instead.

dariomingarelli, 2024-06-18:
> Ctrl-click on the variable or function name to open the declaration.

That works (CMD + click on MacOs).

By the way... I think that is useful also to add it on context menu!

zjanosy, 2024-06-18:
I agree, although once you get used to the Ctrl-click method, it may be even more convenient. But I also miss opening a header file using the context menu, and I did not find how to do this in the new editor.

zjanosy, 2024-06-18:
Actually, Ctrl-click works for opening a header as well...

thaivuongha, 2024-08-08:
Me too. ESP IDE 3.0.0, I pressed F3 or Ctrl+ClickLeft but can not open declaration. All are hidden

kolipakakondal, 2024-08-09:
Hi @thaivuongha Could you please check the below key configured or not? C/C++ Editor in 3.0.0 is LSP based which uses Generic Code Editor keys.
 
[image]

thaivuongha, 2024-08-09:
Hi @kolipakakondal . I tried and succeeded. Thank you so much!

kolipakakondal, 2025-04-10:
Closing this issue as it has been addressed. With CDT-LSP 3.0.0, we also have the "Open Declaration" context menu available.

For Espressif-IDE users, this will be included in the upcoming v3.4.0 release.

[image]