espressif/idf-eclipse-plugin #998: My ESP-IDF v5.2.2 is not the same as the github v5.2.2 tagged version

sfeutrier, 2024-06-20:
**Describe the bug**
I'm trying for hours to make the wifi provisioning sample to work, without any luck!
After investigation of the source code, I found out that everything related to "esp_wifi_start" (starting Wi-Fi in Station Mode) is commented out.
This is not the case here: https://github.com/espressif/esp-idf/blob/v5.2.2/components/wifi_provisioning/src/manager.c (line 1556) while in my folder "C:\Espressif\frameworks\esp-idf-v5.2.2\components\wifi_provisioning\src\manager.c", this is the line 1658... How ?
What is the correct way to solve my issue?

**To Reproduce**
Steps to reproduce the behavior:
I guess install Espressif IDE with esp-idf v5.2.2

**Expected behavior**
Folder files and tagged version files should be identical

**Screenshots**
[image]
[image]

kolipakakondal, 2024-06-21:
Hi @Alvin1Zhang  Could you please point to the right person who can check this?

sfeutrier, 2024-06-24:
Hello,

There is a workaround for this: just go in in the tab "Espressif" and open the "ESP-IDF manager" just install the esp-df version 5.2.2 at another place and select it.
Note: there is an issue that Eclipse Plugging does not succeed to access to the default path to GIT on my side (in the WindowsApp folder), I have just selected another instance at the C root directory

However this is still a real problem as it makes you used a "bugged" version of the ESP-IDF without letting you know at all!

sfeutrier, 2024-10-22:
Any update on this?