espressif/idf-eclipse-plugin #999: Espressif-IDE: Welcome tab overlays everything

jonesman, 2024-06-20:
**Describe the bug**
On my first start of the Espressif IDE, it shows a sort of "Welcome" tab or window. This window overlays any other display element that might open.
So when I go to do the first setup step, open "ESP-IDF Manager", nothing happens. It actually shows up in the background, behind the welcome screen. But especially as a first-time Eclipse user, I don't know that. So I assume immediately something is broken.
I understand the welcome screen is important, and that you can't change Eclipse behaviour, but maybe mention this in the docs ? It caused me major confusion and some lost time of course.

**To Reproduce**
Steps to reproduce the behavior:
Start with a fresh install of Espressif IDE, or remove the Workspace folder

**Expected behavior**
For the "ESP-IDF Manager" to show up

**Espressif-IDE Product Information:**
CDT Build environment variables
PATH: /usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
XAUTHORITY: /run/user/1000/xauth_iZRZAB
INVOCATION_ID: 38f85baa8c1e468d89a6e47847dae2b8
PLASMA_USE_QT_SCALING: 1
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
XDG_CURRENT_DESKTOP: KDE
QT_AUTO_SCREEN_SCALE_FACTOR: 0
JOURNAL_STREAM: 8:4972149
COLORTERM: truecolor
SESSION_MANAGER: local/elga:@/tmp/.ICE-unix/111684,unix/elga:/tmp/.ICE-unix/111684
LOGNAME: jon
KONSOLE_DBUS_SERVICE: :1.27
PROFILEHOME: <NOT FOUND>
MANAGERPID: 111578
PWD: /home/jon/Espressif-IDE
PAM_KWALLET5_LOGIN: /run/user/1000/kwallet5.socket
XCURSOR_THEME: breeze_cursors
QT_WAYLAND_FORCE_DPI: 96
LANGUAGE: en_US:en
WINDOWID: 0
KDE_SESSION_VERSION: 5
XKB_DEFAULT_MODEL: pc105
SHELL: /bin/bash
KONSOLE_DBUS_SESSION: /Sessions/3
OLDPWD: /home/jon
KDE_FULL_SESSION: true
XDG_SESSION_PATH: /org/freedesktop/DisplayManager/Session1
SYSTEMD_EXEC_PID: 111827
COLORFGBG: 15;0
XDG_SESSION_DESKTOP: KDE
LS_COLORS: rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.swp=00;90:*.tmp=00;90:*.dpkg-dist=00;90:*.dpkg-old=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90
SSH_AGENT_LAUNCHER: openssh
SHLVL: 1
KDE_APPLICATIONS_AS_SCOPE: 1
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
KDE_SESSION_UID: 1000
TERM: xterm-256color
XDG_CONFIG_DIRS: /home/jon/.config/kdedefaults:/etc/xdg:/usr/share/desktop-base/kf5-settings
LANG: en_US.UTF-8
XDG_SEAT_PATH: /org/freedesktop/DisplayManager/Seat0
XDG_SESSION_ID: 73
XDG_SESSION_TYPE: wayland
DISPLAY: :1
SHELL_SESSION_ID: 5daeb47b3c824b9da1c6d7c8025e3293
WAYLAND_DISPLAY: wayland-0
KONSOLE_VERSION: 221203
GTK_RC_FILES: /etc/gtk/gtkrc:/home/jon/.gtkrc:/home/jon/.config/gtkrc
XKB_DEFAULT_LAYOUT: de
XDG_SESSION_CLASS: user
_: ./espressif-ide
XCURSOR_SIZE: 24
QTWEBENGINE_DICTIONARIES_PATH: /usr/share/hunspell-bdic/
LIBOVERLAY_SCROLLBAR: 0
GDK_CORE_DEVICE_EVENTS: 1
DESKTOP_SESSION: plasmawayland
USER: jon
GTK2_RC_FILES: /etc/gtk-2.0/gtkrc:/home/jon/.gtkrc-2.0:/home/jon/.config/gtkrc-2.0
QT_ACCESSIBILITY: 1
SSH_AUTH_SOCK: /run/user/1000/openssh_agent
XDG_SEAT: seat0
XDG_RUNTIME_DIR: /run/user/1000
XDG_VTNR: 1
HOME: /home/jon

Operating System: linux
Java Runtime Version: 17.0.11+9
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF version cannot be checked. IDF_PATH or IDF_PYTHON_ENV_PATH  are not set.
Python set for IDF_PYTHON_ENV: <NOT FOUND>

kolipakakondal, 2024-06-21:
Hi @jonesman Thanks for pointing out, will take a look.

jonesman, 2024-06-21:
Thank you for looking into it

alirana01, 2024-09-05:
This will be fixed with the upcoming release

closed 2024-09-05 by alirana01