espressif/idf-eclipse-plugin #1001: Espressif IDE 3.0.0 always freeze in linux

zamek42, 2024-06-22:
**Describe the bug**
new 3.0.0 of Espressif-IDE always freeze under Linux

**To Reproduce**
Steps to reproduce the behavior:
open a header file, or make some modification, which needs code completion, freeze eclipse, and the it restarts. 

**Expected behavior**
normal working

**Espressif-IDE Product Information:**

Operating System: linux
Java Runtime Version: 17.0.11+9
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.4-dev-1030-g0479494e7a
Python set for IDF_PYTHON_ENV: Python 3.11.2

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[hs_err_pid2195613.log](https://github.com/user-attachments/files/15936640/hs_err_pid2195613.log)

kolipakakondal, 2024-06-23:
Hi @zamek42 This is due to the libwebkit2gtk issue with the eclipse as mentioned [here](https://github.com/eclipse-platform/eclipse.platform.swt/issues/843)

Could you set WEBKIT_DISABLE_DMABUF_RENDERER =1 in the environment variable as suggested [here](https://github.com/eclipse-platform/eclipse.platform.swt/issues/843#issuecomment-1802088166) and see if that helps.

zamek42, 2024-06-23:
Hi Kondal Kolipaka,
Thx, it seems work well :)

Indastri, 2024-10-18:
Hi @zamek42, If everything is working can you please close this issue? Thanks!

closed 2024-10-23 by zamek42