espressif/idf-eclipse-plugin #1002: ESPRESSIF IDF manager relies on incorrect python paths

emaayan, 2024-06-24:
when attempting to  install IDF tools in the manager i've often encountered errors when it does:

Executing <path_entered_in_idf_manager>python.exe <tools_path>idf_tools.py export --format=key-value
and i get something like this: 
Executing null/Scripts/python.exe C:\Users\User\eclipse\cpp-2024-06\esp-idf-v5.1.4\tools\idf.py --list-targets
which naturally files with errors of files not found in python 
the problem is that i'm using python i got from msys64, and is in my path. 
so for some reason the IDF manager gets hooked into that. 
there are several problems here:
the first is that this happens even in espressif IDE which is meant to come bundled in 
i would have expected it to take the python exe from it's own variables 

the other issue is that if the python environment is not valid (like missing a scripts folder like what happens in msys64) i would expect the plugin to at least alert that and not continue

alirana01, 2024-06-25:
@emaayan are you using the offline installer or directly downloading IDE from github releases?

emaayan, 2024-06-27:
I'm using the offline installer
, accoriding to the docs, the installer contains an embedded python, and doesn't require anything else to be installed 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

This means that from my pov, that when you add and esp idf framework, the plugin should not only check if the python exe is valid and exist. But also if it's an actually python and not store entry point,  as well if entire folder it resides in complies with python enviorment structure, (for example not us-ng mingw64 python)

alirana01, 2024-08-29:
This was also pointed out internally and the upcoming enhancements should fix this

closed 2024-08-29 by alirana01

emaayan, 2024-12-04:
btw is there a problem using python 3.13 ? as the system python?

flodmotorgrodan, 2025-02-18:
This problem may be related to https://github.com/espressif/idf-eclipse-plugin/issues/1105#issuecomment-2666747708

Where idf_tools.py does not export the path to the ESP-IDF virtual python environment. I made a temporary fix in my id_tools.py file until I understand more.