espressif/idf-eclipse-plugin #1004: Cannot build an imported project that has managed components (undefined reference)

sfeutrier, 2024-06-25:
**Describe the bug**
My team has a project managed on GIT. I got the project and imported it in Espressif-IDE. I first faced an issue saying that my components has been modified locally so I run the bat script to remove the ".component_hash" files.
I tried again to build but now I face an issue of "undefined reference".

`station/managed_components/espressif__iot_bridge/src/bridge_wifi.c:209:(.text.esp_bridge_create_softap_netif+0x18): undefined reference to `esp_netif_create_default_wifi_ap'`

Note it says the calls to esp_netif_create_default_wifi_ap is at line 209 in this file. But it's not. It's at line 285. But I don't know if it's relevant.

**Expected behavior**
I should succeed to build without any issue.

**Espressif-IDE Product Information:**
[Console.txt](https://github.com/user-attachments/files/15967079/Console.txt)

**Eclipse Error log:** 
[image]

kolipakakondal, 2024-06-25:
Hi @sfeutrier Could you please provide a sample project to reproduce the problem so that we can take a look?

sfeutrier, 2024-06-28:
Hello @kolipakakondal 
I cannot provide a sample project as it is under NDA, sorry.
What I can say is that I removed the sdkconfig (not the .default one). Rebuild and tada, it worked!
I have no clue on what happened. It made us this with several Espressif-IDE projects maintained on Git

AnastasiaGrebenkina, 2025-08-16:
> Hello [@kolipakakondal](https://github.com/kolipakakondal) I cannot provide a sample project as it is under NDA, sorry. What I can say is that I removed the sdkconfig (not the .default one). Rebuild [...]

I had the same problem, and I decided to delete sdkconfig. Then I rebuilt the project, compared the old and new (working) sdkconfig files line by line, and in my case it turned out that the old file simply had ‘’# CONFIG_ESP_WIFI_SOFTAP_SUPPORT is not set" (haha :-) ). Replace it with CONFIG_ESP_WIFI_SOFTAP_SUPPORT=y, and everything will work.