espressif/idf-eclipse-plugin #1008: Espressif-IDE 3.0.0 if loggin is enabled on openocd, gdb will not launch

a7v7, 2024-06-27:
**Describe the bug**
OS: Win11
ESP-IDF: 5.2.2
Espressif-IDE: 3.0.0
Target: ESP32-S3-DevKitC1-1 (ESP32-S3-WROOM-2 module)

The standard debug configuration using OpenOCD does not work (that's another defect). One of the things that is requested when reporting a gdb / debug failure is a log file from OpenOCD. I added logging information (-l openocd.log -d2) to OpenOCD config options and ran the debug configuration. I got a log from OpenOCD, but could not get a log from gdb. Closer examination proved that gdb was not launched.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a fresh "ESP-IDF GDB OpenOCD Debugging Configuration (in Debug Configurations)
2. Add -l openocd.log -d2 to the beginning of OpenOCD Setup / Config options.
3. Add set remotelogfile gdb.log to the first line of GDB Client Setup / Commands text box.
4. Run the configuration
5. Since logging is going to a file, there will be no console output.
6. Wait for the timeout.
7. Note that there is a "openocd.log" file, but there is no "gdb.log" file.

**Expected behavior**
I expect that both OpenOCD and gdb will be launched.

alirana01, 2024-09-24:
@a7v7 can you please also attach the log files from the IDE and the openocd.log as well

alirana01, 2024-11-25:
This will be fixed with the 3.1.1 possible issue related to openocd processes hanging

closed 2024-11-25 by alirana01