espressif/idf-eclipse-plugin #1009: Espressif-IDE 3.0.0 cannot launch a debug session.

a7v7, 2024-06-27:
**Describe the bug**
OS: Win11
ESP-IDF: 5.2.2
Espressif-IDE: 3.0.0
Target: ESP32-S3-DevKitC1-1 (ESP32-S3-WROOM-2 module)

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a fresh "ESP-IDF GDB OpenOCD Debugging Configuration (in Debug Configurations)
2. Run the configuration

**Expected behavior**
This should be easy: I am using Espressif-IDE, Espressif ESP-IDF, and an Espressif ESP32-S3-DevKitC1. This is all Espressif tools / hardware. I expect that both OpenOCD and GDB will launch and a debug session will be created.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
ESP-IDF: 5.2.2
Espressif-IDE: 3.0.0
Target: ESP32-S3-DevKitC1-1 (ESP32-S3-WROOM-2 module)

**Eclipse Error log:** 
With some work (see my previous defect) I was able to get an OpenOCD log and two logs from xtensa-esp32s3-elf-gdb.
In the zip file there are three logs:
1. openocd.log -- this is the "d2" log from OpenOCD. It shows that OpenOCD is working perfectly. It is able to communicate with the ESP32-S3 and download code (!!!).
2. gdb_console.log -- this log is copied from the Espressif-IDE console. It shows that gdb cannot access the flash mapping. This log was NOT acquired at the same time as the openocd.log as the debug configuration does not support simultaneous logging from both tools.
3. gdb_gdbinit.log -- this (requisite but useless) log was generated by disabling logging in OpenOCD and enabling logging in gdb. 
[debug.zip](https://github.com/user-attachments/files/16020584/debug.zip)

erhankur, 2024-06-28:
@a7v7 Can you enable the PSRAM support from the menuconfig and give another try?

sigmaaa, 2024-06-28:
Hi @a7v7, a small addition to @erhankur response: to enable PSRAM support in IDE, open sdkconfig file and select "Support for external, SPI-connected RAM" 
[image]

a7v7, 2024-07-01:
My PSRAM support page in the IDE did not look like that. It looked like this:

[image]

I set the interface for Octal SPI, rebuild the application and ran the debug start configuration.

It worked! I really couldn't believe it!

**So why does PSRAM need to be active for this to work?**

erhankur, 2024-07-01:
It is related to OpenOCD stub flasher setting. If PSRAM is not enabled octal flash couldn't be initialized properly. We will fix it in the next OpenOCD release.

closed 2024-07-05 by a7v7