espressif/idf-eclipse-plugin #1010: Espressif-IDE 3.0.0 "Product Information" returns File Not Found error

LucasSaintrain, 2024-06-28:
**Describe the bug**
I've just made a new clean installation of the new 3.0 IDE.
So far there's been no issues with the IDE except for the output of this particular command.

The 'Espressif > Product Information' command outputs correct information up until the IDF Eclipse Plugin Version,
then it outputs the traceback for a 'FileNotFoundError'.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Espressif > Product Information'
2. See error

**Expected behavior**
A normal output without any errors.

**Espressif-IDE Product Information:**
[Product Information.txt](https://github.com/user-attachments/files/16025197/Product.Information.txt)

pedrominatel, 2024-06-28:
Works fine on macOS:

```
Operating System: mac os x
Java Runtime Version: 17.0.11+9
Eclipse Version: 4.30.0.v20231201-0110
Eclipse CDT Version: 11.4.0.202309142347
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.4-dev-1030-g0479494e7a
Python set for IDF_PYTHON_ENV: Python 3.9.6
```

LucasSaintrain, 2024-06-30:
I forgot to mention, I'm using Windows 11.