espressif/idf-eclipse-plugin #1011: Espressif-IDE 3.0.0 can't change Eclipse's Code Formatter Profile

LucasSaintrain, 2024-06-30:
**Describe the bug**
Eclipse by default does code formatting using tabs, while  esp-idf examples use 4 spaces.

To change this, I went to Window>Preferences then C/C++>Code Style>Formatter to change my active profile.
I created a new profile (based on 'BSD/Allman [built-in]') and changed 'Tab policy' from 'Tabs only' to 'Spaces only'.

The profile preview in formatter preferences correctly showcases spaces instead of tabs.
However, in my C code this doesn't change anything, even after restarting the IDE and even the computer.

There isn't any Project Specific Setting, this error happens for any project.

This error is annoying since auto indentation is done by eclipse using tabs and I have to manually erase and insert spaces.

Note: I've recently cleanly installed the new IDE (yesterday).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Window>Preferences'
2. Go to 'C/C++>Code Style>Formatter'
3. Select or a create a profile that uses a 'Spaces only' 'Tab policy'.
4. Go back to C code and insert a tab anywhere.
5. View a tab instead of spaces (error)

**Expected behavior**
Four spaces inserted when I press 'tab' and when auto indentation occurs.
Also, I think a 'Spaces Only' policy should be enabled by default since that's the norm for IDF projects.

**Espressif-IDE Product Information:**
[Product Information.txt](https://github.com/user-attachments/files/16042965/Product.Information.txt)

Note: there is a strange error at the end of my product information, that I reported here: #1010 .
This error seems unrelated to the current problem, though.

**Exported Code Formatter Profile:** 
[Product Information.txt](https://github.com/user-attachments/files/16042966/Product.Information.txt)

LucasSaintrain, 2024-06-30:
Actually, I have a sneaking suspicion that my problem is much bigger and encompasses most of my Eclipse's Preferences.

I've tried using doxygen in 'C/C++>Editor' then 'Workspace Default'='Doxygen', but this never worked.
I've disabled automatically closing of strings, parenthesis, brackets and braces. This change didn't take effect.
I've enabled folding of preprocessor branches, and it also didn't work.

The only change I did that ever worked was creating a new C source file template and a new C header file template.

viteo, 2024-07-01:
have a look #987 at my journey through

LucasSaintrain, 2024-07-01:
@viteo thank you, that helps a lot!

LucasSaintrain, 2024-07-02:
So as far as I understand 'C/C++ > Editor (LSP)' option 'Set C/C++ Editor (LSP) as default' should work for enabling/disabling clangd.
In reality it doesn't do anything, and clangd is always used.
This, in turn, renders all  'C/C++ > Editor' options useless.

kolipakakondal, 2024-07-02:
Hi @LucasSaintrain We are sorry if things are not very clear for you.

> So as far as I understand 'C/C++ > Editor (LSP)' option 'Set C/C++ Editor (LSP) as default' should work for enabling/disabling clangd.
In reality it doesn't do anything, and clangd is always used.
This, in turn, renders all 'C/C++ > Editor' options useless.

C/C++ Editor (LSP) is the default and only the editor supported with the Espressif-IDE for esp-idf projects.

I have updated about clangd here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.0.0

> This includes the [Eclipse CDT-LSP](https://github.com/eclipse-cdt/cdt-lsp/) plugins, enabling support for the latest C/C++ standards and providing an LSP-based C/C++ Editor. This editor, powered by the [LLVM](https://github.com/espressif/llvm-project) clangd C/C++ language server(clangd distributed along with the esp-clang toolchain with [esp-17.0.1_20240419](https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240419) and higher), offers advanced functionality for ESP-IDF developers.

> In line with this enhancement, we've discontinued support for the standard CDT Editor/Indexer, as it only offers support for up to C++ 14. This has been replaced with a new LSP editor, especially considering that ESP-IDF now utilizes C++ 20 (with GCC 11.2) in v5.0.x, transitions to C++ 23 (with GCC 12.1) in v5.1, and to C++ 23 with GCC 13.1 in v5.2

> The LSP powered C/C++ editor greatly benefits ESP-IDF developers by aligning with the latest language standards and compiler versions, enhancing productivity, and improving code quality.

We have kept standard C/C++ editor options and preferences options as-is considering some developers want to use them for non-esp based projects. Probably, we could optimise these options going forward by introducing the new ESP-IDF Perspective in the IDE.

kolipakakondal, 2024-07-02:
Regarding the code formator usage with the clangd, could you please check this post here https://www.esp32.com/viewtopic.php?f=40&t=40263&sid=af321e93aa5b026a47e02b1904b462bd#p133952

>  Here is the way to generate default .clang-format file based on the [configurable format style options](https://clang.llvm.org/docs/ClangFormatStyleOptions.html#configurable-format-style-options)

> For example in macOS:
> $ brew install clang-format
> $ clang-format -style= GNU -dump-config > .clang-format

> Then one could customize the .clang-format file based on their preferences.

LucasSaintrain, 2024-07-02:
Hi @kolipakakondal, thank you. Your answers helped me find some directions.

I'm on windows, so I actually used ESP-IDF Terminal through the IDE to run the clang-format command:
`clang-format -style=GNU -dump-config > "C:\Espressif\workspace\[PROJECT FOLDER]\.clang-format"`
This correctly generated the .clang-format at the root of my project (it doesn't show in the IDE's explorer, but I think that's normal).

Now, even with the clang-format file, tabs are still inserted when I press tab.
I realized that what I wanted is actually an option in 'General > Editors > Text Editors' 'Insert spaces for tabs'.

Also since I now have a configured clang format, I'm able to enable 'Editor (LSP) > Save action' to correctly format the source file after each save.

Thanks for helping me solve this!

closed 2024-07-02 by LucasSaintrain

DigKleppe, 2024-07-06:
Hi @kolipakakondal,
I managed to get  my linewidth customized in the clang-format. Added the clang plugin in Eclipse.  Format source with the source menu -> format. But i did not succeed in disabling the managed-components folder. Tried with .clang-format-ignore and a .clang-format file holding "DisableFormat: true SortIncludes: Never" in the managed-compontents folder.
Can you point me to a way to do this? I have included a small project (hello) that formats ok.
My real project give me an error " AST not found" . I included the logfile.
Thanks.
[error.log](https://github.com/user-attachments/files/16116051/error.log)
[hello.zip](https://github.com/user-attachments/files/16116052/hello.zip)

kolipakakondal, 2024-07-08:
> Hi @kolipakakondal, I managed to get my linewidth customized in the clang-format. Added the clang plugin in Eclipse. Format source with the source menu -> format. But i did not succeed in disabling [...]

Hi @DigKleppe I wasn't sure why the .clang-format-ignore approach wasn't working, but I found that adding the following flags in the `.clang-format` file in the `managed-components` root works for me. Please see if this is helpful for you.

```
DisableFormat: true
```

Here is the project structure
```
project
 - managed_components
        - .clang-format  
 - main
 - .clang-format  
```

The flag `DisableFormat: true` tells ClangFormat to completely ignore this specific `.clang-format` file and its formatting rules within the managed-components directory.

kolipakakondal, 2024-07-08:
@DigKleppe Is it possible for you to provide a sample project which has `AST not found` formatting exception so that we can troubleshoot further.

DigKleppe, 2024-07-09:
I have got the AST-not-found error solved by a complete rebuild. If I select a file in the project explorer the formatting works OK. Only via the menu, not via the shortkeys, despite the setting of the code formatter to cppstyle.  I can live with that.  Formatting the complete project keeps formatting all folders. The disable clang file is not seen. 
Thanks !
Dig