espressif/idf-eclipse-plugin #1014: use flash monitor in eclipse failed (IDFGH-13176)

ArvinHou, 2024-07-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

the setting in eclpse is bellow. Add "monitor" in launch configuration.
[image]
when use IDF v4.4.5, It works ok.When I click run, it automatically downloads and starts monitoring。
When use IDF v5.2.2,It works  with error.
```
error: Monitor requires standard input to be attached to TTY
```
[image]
How to solve this problem?
I know I can manually open the terminal。But I'm still used to the kind of automatic monitoring。
Can it ？
[image]

kolipakakondal, 2024-07-02:
Hi @ArvinHou Thanks for your ticket. Currently it's not possible to open the monitor automatically after the flash. We are working on this enhancement, it will be part of the next release.
Here is the PR https://github.com/espressif/idf-eclipse-plugin/pull/1013

Here is the update site archive for this PR
https://github.com/espressif/idf-eclipse-plugin/actions/runs/9744907990/artifacts/1655560411

Please try out and see how it goes!

ArvinHou, 2024-07-03:
@kolipakakondal  Thanks，I have tested and provided feedback.

closed 2024-07-19 by kolipakakondal