espressif/idf-eclipse-plugin #1019: missing code outline, jumping to declarations and includes etc.

bs-eng, 2024-07-11:
Hello!

**Describe the bug**
The eclipse environment is missing the following functionalities after installing IDF-Eclipse-PluginV3.0.0:
- code outline of currently opened file in outline tab
- jumping to declarations of types, variables, functions when double clicking them
- jumping to included files when double clicking the #include statement
Am I missing something in eclipse or is it simply a bug of the plugin?

**To Reproduce**
Steps to reproduce the behavior:
1. install eclipse IDE 2024-03
2. install IDF-Eclipse-Plugin
3. open project
4. open any code file

**Expected behavior**
Above functionalities should work as before V3.x

**Screenshots**
[image]

**Espressif-IDE Product Information:**
[ESP32_tools.txt](https://github.com/user-attachments/files/16179651/ESP32_tools.txt)

**Eclipse Error log:** 
[eclipse_error_log.log](https://github.com/user-attachments/files/16179652/eclipse_error_log.log)

Thanks for your help!
Regards
JR

kolipakakondal, 2024-07-11:
Hi @bs-eng With Espressif-IDE 3.0.0, we've switched to a [Clangd-based C/C++ editor](https://github.com/eclipse-cdt/cdt-lsp). However, this is only supported for ESP-IDF version 5.1 or later. Unfortunately, I don't have an immediate solution for using it with earlier versions. I'll check into it further and get back to you.

In the meantime, could you please stick with the previous version, [v2.12.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.1)?

bs-eng, 2024-07-11:
Hi!
I would happily stick with a previous version if I would find any hint of how to install that. All install guides that I found so far give instructions for installing latest version only and no clue as to how to change the url that points to the corrected sources.
So far my manual attempts at doing so failed.
The downloadable zip file at the link you gave, contains lots of files, which are unclear to me what I need to to with them in order to get them installed in eclipse.
If you could help me out on that, I would really appreciate it!

Do you happen to know when ESP32-ADF is scheduled to support IDFv5.1?

Cheers
JR

kolipakakondal, 2024-07-12:
> Hi! I would happily stick with a previous version if I would find any hint of how to install that. All install guides that I found so far give instructions for installing latest version only and no [...]

Ok, two options:

https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.12.1-win32.win32.x86_64.zip

or
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-2.12.1.zip 
Here are the instructions for installation https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#installing-idf-eclipse-plugin-from-local-archive

bs-eng, 2024-07-13:
Dear @kolipakakondal,
> Here is what you could try without leaving 3.0.0 version:
> 
> 1. Download latest esp-clang toolchain from here https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240419
> 2. Configure `clangd` path in the preferences as shown below and click on apply and close.
> Now see if your're able use clangd editor features.

I guessed I need to download both lib and clang itself from the linked location. I tried in different ways to 'install' this, but found no effective way of doing it. Tried the "install new software" in eclipse as well as just copying it to a folder and pointing to it. Neither worked.

So I am setting up my system anew using the downloadable 2.12.1 version.
Thanks for the link to the old version! :-)
JR

bs-eng, 2024-07-13:
For those stumbling upon this, here is how I finally installed the previous version:

1. download the version of _Espressif IDF Plugins for Eclipse_ of your choice (dont get irritated by the word _update_ in the name, it is a full package)
2. in Eclipse uninstall the previous version of IDF plugin:
     - go to eclipse > help > about eclipse
     - click installation details
     - type in search bar _ESP_
     - select the IDF plugin and uninstall
     - follow eclipse instructions
3. Eclipse should ask to restart Eclipse to finish
4. quit Eclipse
5. delete Espressif tools that were installed previously, as they do not get deleted during the uninstall step before!
     - most probably in your user folder (on Windows) there is the .espressif folder, delete it completely!
     - unix/linux/mac I do not know, please search on internet
6. in Eclipse install the new IDF plugin according to [standard instructions](https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#installing-idf-eclipse-plugin-from-local-archive) as posted by @kolipakakondal 
7. install IDF tools as usual: Eclipse > Espressif > ESP IDF tools manager > install tools

Opening projects afterwards:
I deleted the projects in my workspace and simply imported them again, to make sure everything gets set properly in the .cproject file (which I deleted before importing)

That's how it worked for me at least.
Cheers
JR

kolipakakondal, 2024-07-15:
> Dear @kolipakakondal,
>
>
> I guessed I need to download both lib and clang itself from the linked location. I tried in different ways to 'install' this, but found no effective way of doing it. Tried [...]

Okay, here's what I did on my macOS (arm64) system, and it worked without any problems. If you'd like to try it out later, feel free to do so.

1. Installed ESP-IDF v4.4.8

[image]

2. Downloaded my OS supported esp-clang ([clang-esp-17.0.1_20240419-aarch64-apple-darwin.tar.xz](https://github.com/espressif/llvm-project/releases/download/esp-17.0.1_20240419/clang-esp-17.0.1_20240419-aarch64-apple-darwin.tar.xz)) from [here](https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240419)

3.  Changed clangd path in the preferences.
[image]

5. Compiled the blink example for esp32, no build issues reported.

[image]

7. Opened the C/C++ editor and which is by default opened in the LSP editor and working fine.

https://github.com/user-attachments/assets/d7a1bfd3-37ef-4b57-b666-5f02176b3e51