espressif/idf-eclipse-plugin #1024: Add search functionality to ESP-IDF serial terminal monitor

emaayan, 2024-07-26:
it would be a huge help it were possible to have a simple find window in side the serial terminal to search for specific log entries. 
just press either Ctrl+F or have a button in the terminal toolbar (or both) that pops up a window to search for a string to jump to that place.

kolipakakondal, 2024-07-29:
Hi @emaayan Thanks for your request. We could think of implementing something similar to console search

[image]

emaayan, 2024-07-29:
> Hi @emaayan Thanks for your request. We could think of implementing something similar to console search
> 
> [image]
> 

Exactly ,sometimes when compiling and i look for something, it's easy to miss
2 other suggestions for the serial monitor , almost all of my api call are accompanied with esp_error_check however when i fail and it goes into a boot loop, it super hard to stop the monitor from scrolling so fast to actually see the error, not sure if it's possible to tell the monitor, maybe i just need to disconnect it.

I've noticed the serial monitor has hyperlinks to the tags, but those aren't very conclusive as they always assume it's a resource name, i'm not sure what's the convention for tags in logs, but in theory i think you could use the absract syntax tree to know what calls to logs use what tags, so if one hyper links on a tag it would be offered only files that actually use said tag.

kolipakakondal, 2024-07-29:
By the way, we have a simple search filter available when launching the serial monitor. You can specify the text you want to filter out. If you know the exact tag/text you are looking for, you can use this feature for now.

[image]