espressif/idf-eclipse-plugin #1026: Debugging ESP32S3 in Eclipse IDE

jrdecesare, 2024-07-30:
Hi!
For quite some time I was able to compile and debug my ESP32S3 project.  I went back to the project a  few weeks later and I am now getting this error message when I debug:

Usage: idf.py Marine\BMS\Software\BMS_Test_App\build [OPTIONS]
Try 'idf.py Marine\BMS\Software\BMS_Test_App\build --help' for help.

Error: No such option: -p

It seems as if the -p is no longer an option??   I have my USB connected to my workstation as COM9.  This used to work so I don't know if a windows update had something to do with this.

I also:
* downloaded the latest tools ESP-IDF-V5.3
* can compile the code succesfully
* deleted the Debug Configuration and recreated one. Problem still exists.
* Rebooted my workstation twice.

Thanks!!
John

[image]
[image]

sigmaaa, 2024-07-31:
Hi @jrdecesare,

I could not reproduce this issue with the ESP32-S3 devkit and ESP-IDF 5.3 (I've tested the uart flashing, the serial monitor, and the openocd debugging). It's weird, that you're encountering this problem when trying to debug, as the -p argument is used for UART flashing. If you use the debug configuration (ESP-IDF GDB OpenOCD Debugging), you don't need the UART port, as flashing is done via JTAG.

If you are using the ESP32-S3 devkit, please follow these steps for debugging:

1. Create the ESP-IDF GDB OpenOCD Debugging configuration if you don't already have one.

2. Edit the Configuration and navigate to the Debugger tab -> Select the esp32s3 target and ESP32-S3 chip (via built-in USB-JTAG) board.

2. Specify Flash Size on Windows: Sometimes it's necessary to specify flash size on Windows. To do it: Open sdkconfig in your project (double-click the file). Search for Serial flasher config and set the Flash size to 4MB.

3. Connect the Device: The ESP32-S3 has two micro USB ports (USB and UART). Connect to the USB port.

4. Start the Debug Session: Select the debug configuration in debug mode and click "Debug". The flash process will be done automatically via JTAG, and then the debug session should start.

jrdecesare, 2024-07-31:
Hi Denys,
Thanks for the quick reply!  I actually downloaded the latest Eclipse (2024 06) and via their website, dragged over the ESP-IDF Eclipse Plugin which is here:

https://marketplace.eclipse.org/content/esp-idf-eclipse-plugin

I can build my application fine with no errors.

***@***.***

I created a debug configuration as such:

***@***.***

***@***.***

When I debug, I get this error:

***@***.***

Any help would be appreciated as I’ve spent about 5 hours trying to get this to work.

Thanks!

John

  3.  Specify Flash Size on Windows: Sometimes it's necessary to specify flash size on Windows. To do it: Open sdkconfig in your project (double-click the file). Search for Serial flasher config and set the Flash size to 4MB.
  4.  Connect the Device: The ESP32-S3 has two micro USB ports (USB and UART). Connect to the USB port.
  5.  Start the Debug Session: Select the debug configuration in debug mode and click "Debug". The flash process will be done automatically via JTAG, and then the debug session should start.

sigmaaa, 2024-07-31:
> I can build my application fine with no errors.
> 
> ***@***.***
> 
> I created a debug configuration as such:
> 
> ***@***.***
> 
> ***@***.***
> 
> When I debug, I get this error:
> 
> ***@***.***

Hi @jrdecesare,

we can't see the screenshots. Could you please upload it again? Thank you

jrdecesare, 2024-07-31:
Hi Denys,

I have attached them in a word document. Sorry about that!!

Thanks!

John

sigmaaa, 2024-08-01:
Hi @jrdecesare 

we can't see the word document either. Looks like github blocking the files uploaded from the email

jrdecesare, 2024-08-01:
Then how should I get you the files?  Do you have another non github address?

Thanks!

John

sigmaaa, 2024-08-02:
Hi @jrdecesare,

you can send them via a browser on the GitHub page or to me at this address - almazovdenys@gmail.com. Thanks

sigmaaa, 2024-08-02:
Hi @jrdecesare,

the configuration that you are showing on the screenshot (ESP-IDF Application) is for flashing. For openOCD debugging, you must use the "ESP-IDF GDB OpenOCD Debugging" configuration. I've deleted the default one and recorded a small demo of how the debug should work. Please, take a look:

https://github.com/user-attachments/assets/1bbbfc69-8fd5-43ba-8ab2-944155a6bdbe

PS I'm using the latest master, so the icon for the configuration is different

sigmaaa, 2024-08-02:
Hi @jrdecesare,

one more thing. It's not clear why you are having this issue:
[image]
Could you please provide an error log after you face it again by following these steps? https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#error-log

sigmaaa, 2024-08-07:
Hi @jrdecesare,

Is there any update on this? Were you able to debug?

jrdecesare, 2024-08-07:
Hey Denys,
I have not yet.  I am actually on vacation until next week. Thank you very much for keeping in touch. I will get back on this when I return late next week. Again, thanks!!! Most appreciated.

John

sigmaaa, 2024-09-12:
Hi @jrdecesare,

Is there any update on this?

closed 2024-10-03 by sigmaaa