espressif/idf-eclipse-plugin #1036: Cannot open declaration

MrDuby, 2024-08-25:
**Describe the bug**
I installed the Eclipse base ide, Espressif IDE v3.0.0 on windows 10 professional. When I start a project or import any project I have no way to open function or variable declaration. On right click I cannot even find the option "Open Declaration" in the menu. Tried CTRL+click but I'm not even getting the Hyperlink to click to find declaration. When I select lets say the function and press F3 I get a pop-up message saying "The operation is not applicable to the current selection. Select a hyperlink target". The code compiles correctly though.

kolipakakondal, 2024-08-26:
Hi @MrDuby Could you please check this https://github.com/espressif/idf-eclipse-plugin/issues/995#issuecomment-2276935107

amin352, 2024-08-26:
I have the same problem and I do many ways to solve the problem include the way you said in other topic, but the problem still remain.
I can't open declaration in any way and the open declaration in Navigate tab also is not usable and is off
please help

MrDuby, 2024-08-27:
This is how I finally solved it.
I downloaded the offline installer(Espressif-IDE 3.0.0 with ESP-IDF v5.2.2) and no more the online installer(Universal Online Installer 2.28). After the installation I had to also go to IDE toolbar->Espessif->ESP-IDF Manager then Add ESP-IDF
[image]

amin352, 2024-08-28:
> This is how I finally solved it. I downloaded the offline installer(Espressif-IDE 3.0.0 with ESP-IDF v5.2.2) and no more the online installer(Universal Online Installer 2.28). After the installation [...]

Thanks for your solution
I apply it and it works finally, BUUUUUT after closing ide again the problem still exist and not solving either

JosephKissell, 2024-11-19:
None of these solutions work for me.

kolipakakondal, 2024-11-19:
Hi @Ignisuti  Could you please share an error log and LSP log?

Rrror log https://preview-docs.espressif.com/projects/espressif-ide/en/latest/troubleshooting.html?#error-log

LSP logs can be enabled from the preferences.

[image]

Ignisuti, 2024-11-19:
I'm not sure how to send the log. This is what I have in my settings. However, there are no files at this location. I've restarted Eclipsed and re-compiled.

[image]

kolipakakondal, 2024-11-19:
> I'm not sure how to send the log. This is what I have in my settings. However, there are no files at this location. I've restarted Eclipsed and re-compiled.
> 
> ![Screenshot 2024-11-19 001238](https://private-user-images.githubusercontent.com/37918015/387521304-1143c906-ae1b-4259-888d-1e714d187884.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5OTc1NTMsIm5iZiI6MTczMTk5NzI1MywicGF0aCI6Ii8zNzkxODAxNS8zODc1MjEzMDQtMTE0M2M5MDYtYWUxYi00MjU5LTg4OGQtMWU3MTRkMTg3ODg0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDA2MjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0ZDJjY2ZiZmI3OWQ2ZjcxNGUzZDZhOThlNDU1ZmZhNDg3NWM1YTVhYmYyOWFlZjkwYTI3NDIwODZkZDRjNjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SL5hUKED3QN6CK532TFi-bgwL8Syqt8EwB2ayO6fv6Y)

Please navigate to the location mentioned in the dialog. 

ex:
/Users/myuser/workspace-devcon/languageServers-log/org.eclipse.cdt.lsp.server.log

ua100k, 2024-11-22:
Same problem here. with 3.0. Previous versions have no issues.

Ignisuti, 2024-11-25:
I do not have the Language Servers option in my console. 
[image]

I'm not familiar with clangd, but based on what you have said, I believe mine is setup correctly.
[image]

sigmaaa, 2024-11-29:
Hi @Ignisuti,

would it be possible to connect on MS Teams to work through this issue together? The problem is quite unusual, and we have not been able to reproduce it on our end. If this works for you, please write to [ide.support@espressif.com](mailto:ide.support@espressif.com) with your time zone and a suitable time. We will then send you an invite.

Best regards,
Denys

kolipakakondal, 2024-11-29:
> Error log https://preview-docs.espressif.com/projects/espressif-ide/en/latest/troubleshooting.html?#error-log

Can you please share an error log?

Most likely, the clangd server might be failing during processing.

Ignisuti, 2024-12-02:
Update...
I have worked with Denys and he discovered the following solution:

1.) I had turned on my Indexer during troubleshoot. It needed to be re-disabled.
2.) These 3 options for the LSP Editor were NOT enabled by default. Enabling these fixed my issue.
[image]

ua100k, 2024-12-02:
This has no effect on my system. The language were enabled already. In previous editions, the index was enabled and no option for language servers are available in preferences. Not sure if that is helpful.

sigmaaa, 2024-12-02:
> This has no effect on my system. The language were enabled already. In previous editions, the index was enabled and no option for language servers are available in preferences. Not sure if that is helpful.

Hi @ua100k,

The LSP editor was included to the IDE in the version 3.0.0 and higher and was intended to replace default C/C++ editor. Could you share what your preference is for clangd? Like here
[image]

Also, make sure that you are opening c/c++ files with LSP editor (right-click on the file -> open with C/C++ Editor (LSP)). Or just set this in the preferences like this:
[image]

Please note that after building the project, the C/C++ Language Server should appear as running in the bottom-right corner for a short period. Once this process completes, you can use Ctrl + Left Click (default) to open declarations.

[image]

ua100k, 2024-12-02:
Hi @sigmaaa 

Thank you for your quick reply and help.

Here is my clangd config, Editor Config and Language server showing.  I don't have this issue with the same file in IDE 2.9. 
[image]
[image]

[image]

sigmaaa, 2024-12-03:
Hi @ua100k,

The settings look good to me. Are you able to open declarations in the hello_world project? We can also troubleshoot this problem in Teams. You can write to [ide.support@espressif.com](mailto:ide.support@espressif.com) with your time zone and a suitable time, and we will send you an invite.

ua100k, 2024-12-10:
@sigmaaa I wrote to ide.support but did not get any reply or call on the date and time suggested. Can you please advise. 

Thank you

sigmaaa, 2024-12-10:
Hi @ua100k,

I apologize for missing your earlier email. I've replied and suggested the same time for tomorrow. Please let me know if that works for you. Thank you!

DigKleppe, 2024-12-29:
This issue is present a long time now. When a file is opened with the old (none-LSP) C editor F3 works. But you have to do this every time, even if  you checked the open-always-with checkbox.  Could this be fixed to use the old editor by default?
Thanks.

sigmaaa, 2025-07-01:
> Same issue still present with the version 3.5.0 .
> 
> Actually, everthing is fine the very first time(s?) the editor is used. After some days the message "The operation is not applicable to the current selection. Select a hyperlink target." pops-up trying to open the declation of a struct instance.

Hi @RELOC-DEV,

Is the language server running at that moment?
You can check this by opening the "Language Servers" view (via Window > Show View > Other... > Language Servers) and verifying whether the server is active. Also, look at the status bar or the "Error Log" view for any issues related to the language server.

RELOC-DEV, 2025-07-09:
Hi @sigmaaa,
sorry for my late reply but, obviously, the editor did not show the issue for a while.
Now the problem is present again but in a more underhanded way. I mean, trying to open a function declaration the message appears:

[image]
(note that the Language Server is running)

But when I try again few seconds later the editor correctly shows the declaration.

Another sign is that the "Outline" window is empty (for this specific file only) when this happens, closing and reopening the file makes the "Outline" windows to show the content correctly.