espressif/idf-eclipse-plugin #1040: ESP-IDF Manager: 'Installing tools...' has encountered a problem. Cannot invoke "String.indexOf(int)" because "value" is null

Abdull, 2024-08-27:
**Describe the bug**
In a freshly installed *Espressif IDE 3.0.0* (via `Espressif-IDE-3.0.0-linux.gtk.x86_64.tar.gz`), despite fulfilling all prerequisites and following the steps installations steps found in https://github.com/espressif/idf-eclipse-plugin/blob/27f07351f83acc220e4630d29120d2dd89a02c07/README.md#Installation , I cannot setup the ESP-IDF via the ESP-IDF Manager.

[image]

Neither "*Use an existing ESP-IDF directory from file system*" (providing existing ESP-IDF directory `/opt/install/esp-idf`) nor "*Download ESP-IDF*" work. Both will eventually result in a "Problem Occured" modal with the error message:

> 'Installing tools...' has encountered a problem.
> Cannot invoke "String.indexOf(int)" because "value" is null

[image]

The Eclipse *Console* view has the following output:

<details>
  <summary>Eclipse *Console* output (click to expand)</summary>

```
Installing tools...
This can take a while. Please be patient.
Executing /usr/bin/python3.11 /opt/install/esp-idf/tools/idf_tools.py install all
Selected targets are: esp32, esp32c2, esp32p4, esp32s3, esp32h2, esp32c6, esp32c3, esp32c5, esp32s2, esp32c61
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, esp-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, esp-rom-elfs, qemu-xtensa, qemu-riscv32
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping esp-clang@esp-17.0.1_20240419 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20240726 (already installed)
Skipping ninja@1.11.1 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Skipping qemu-xtensa@esp_develop_9.0.0_20240606 (already installed)
Skipping qemu-riscv32@esp_develop_9.0.0_20240606 (already installed)

Executing /usr/bin/python3.11 /opt/install/esp-idf/tools/idf_tools.py install-python-env
Python 3.11.2
pip 24.2 from /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
Skipping the download of /home/abdull/.espressif/espidf.constraints.v5.4.txt because it was downloaded recently.
Upgrading pip and setuptools...
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: pip in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (24.2)
Requirement already satisfied: setuptools in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (71.0.0)
Installing Python packages
 Constraint file: /home/abdull/.espressif/espidf.constraints.v5.4.txt
 Requirement files:
  - /opt/install/esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 3)) (71.0.0)
Requirement already satisfied: packaging in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 4)) (24.1)
Requirement already satisfied: click in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 7)) (8.1.7)
Requirement already satisfied: pyserial in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 8)) (3.5)
Requirement already satisfied: cryptography in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 9)) (43.0.0)
Requirement already satisfied: pyparsing in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 10)) (3.1.4)
Requirement already satisfied: pyelftools in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 11)) (0.31)
Requirement already satisfied: idf-component-manager in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 12)) (2.0.3)
Requirement already satisfied: esp-coredump in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.11.0)
Requirement already satisfied: esptool in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 14)) (4.8.dev5)
Requirement already satisfied: esp-idf-kconfig in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 15)) (2.3.0)
Requirement already satisfied: esp-idf-monitor in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 16)) (1.4.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 17)) (0.1.2)
Requirement already satisfied: esp-idf-size in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 18)) (1.5.0)
Requirement already satisfied: esp-idf-panic-decoder in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 19)) (1.1.0)
Requirement already satisfied: pyclang in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 20)) (0.4.2)
Requirement already satisfied: construct in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 21)) (2.10.70)
Requirement already satisfied: freertos_gdb in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from -r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 24)) (1.0.3)
Requirement already satisfied: cffi>=1.12 in /home/abdull/.espressif/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages (from cryptography->-r /opt/install/esp-idf/tools/requirements/requirements.core.txt (line 9)) (1.17.0)
[... 139 lines cut ...]
	at java.base/java.lang.ProcessEnvironment$Value.valueOf(ProcessEnvironment.java:203)
	at java.base/java.lang.ProcessEnvironment$StringEnvironment.put(ProcessEnvironment.java:243)
	at java.base/java.lang.ProcessEnvironment$StringEnvironment.put(ProcessEnvironment.java:221)
	at java.base/java.util.AbstractMap.putAll(AbstractMap.java:281)
	at com.espressif.idf.core.ProcessBuilderFactory.run(ProcessBuilderFactory.java:38)
	at com.espressif.idf.ui.tools.ToolsJob.runCommandIdfPyInIdfEnv(ToolsJob.java:359)
	at com.espressif.idf.ui.tools.ToolsJob.loadTargetsAvailableFromIdfInCurrentToolSet(ToolsJob.java:164)
	at com.espressif.idf.ui.tools.ToolsInstallationJob.run(ToolsInstallationJob.java:103)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

!ENTRY com.espressif.idf.core 4 0 2024-08-27 19:26:46.968
!MESSAGE Cannot invoke "String.indexOf(int)" because "value" is null
!STACK 0
java.lang.NullPointerException: Cannot invoke "String.indexOf(int)" because "value" is null
	at java.base/java.lang.ProcessEnvironment.validateValue(ProcessEnvironment.java:120)
[... cut ...]
	at java.base/java.lang.ProcessEnvironment$StringEnvironment.put(ProcessEnvironment.java:221)
	at java.base/java.util.AbstractMap.putAll(AbstractMap.java:281)
	at com.espressif.idf.core.ProcessBuilderFactory.run(ProcessBuilderFactory.java:38)
	at com.espressif.idf.ui.tools.ToolsJob.runCommandIdfPyInIdfEnv(ToolsJob.java:359)
	at com.espressif.idf.ui.tools.ToolsJob.loadTargetsAvailableFromIdfInCurrentToolSet(ToolsJob.java:164)
	at com.espressif.idf.ui.tools.ToolsInstallationJob.run(ToolsInstallationJob.java:103)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
```

Abdull, 2024-08-30:
OP here. I can happily report that this issue fixed itself. When I just retried adding ESP-IDFs in my Espressif IDE installation, it just worked.

Both "use an existing ESP-IDF directory from file system" and "download ESP-IDF" worked.

The only things that have changed or were done *from my side* since I reported this issue were reboots.

* Tested successfully with *all* ESP-IDF plugins at version `1.0.1.202408211148`. 
* Also tested successfully with *ESP-IDF SDK Config Core* (`com.espressif.idf.sdk.config.core`) at version `1.0.2.202408291042`.

I therefore close this issue. Feel free to reopen it if needed.

closed 2024-08-30 by Abdull

lbthomsen, 2024-12-03:
I am still seeing this issue with latest versions everywhere.

komcconnell, 2024-12-05:
Happening with me too. (Mac OSX Sonoma).

alirana01, 2024-12-05:
Hi @komcconnell can you test it with the latest nightly build of the espressif-ide. This is planned for the upcoming release the nightly should contain the fix for this.

lbthomsen, 2024-12-05:
> Hi @komcconnell can you test it with the latest nightly build of the espressif-ide. This is planned for the upcoming release the nightly should contain the fix for this.

See my comment in #1073 - nightly build as of a few days ago - still failing.

Manikanta25055, 2025-01-09:
me too, I am also facing the same issue

flodmotorgrodan, 2025-02-18:
This problem may be related to https://github.com/espressif/idf-eclipse-plugin/issues/1105#issuecomment-2666747708

Where idf_tools.py does not export the path to the ESP-IDF virtual python environment.

rtek1000, 2025-06-29:
I just got this notice. Shouldn't it have been resolved?