espressif/idf-eclipse-plugin #1042: Espressif-IDE (v3.0.0) unable to 'Install New Software...': An error occurred while collecting items to be installed

carlessole, 2024-08-29:
Espressif-IDE (v3.0.0): Installing a Plugin through 'Install New Software...', after pressing 'Finish' following error appears and plugin is not installed:

> An error occurred while collecting items to be installed
> session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
> Cannot invoke "java.lang.Comparable.compareTo(Object)" because the return value of "java.util.function.Function.apply(Object)" is null
> Cannot invoke "java.lang.Comparable.compareTo(Object)" because the return value of "java.util.function.Function.apply(Object)" is null

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Help -> Install New Software...'
2. Select the plugin you want to install and follow the steps until pressing the 'Finish' button
3. Installation fails. See error

**Exactly the SAME procedure in Espressif-IDE v2.8.1 succeeds and plugin is installed properly.**

**Expected behavior**
To be able to install plugins through 'Install New Software...'

**Screenshots**
[image]

**Espressif-IDE Product Information:**
> Espressif-IDE is an Integrated Development Environment(IDE) based on Eclipse CDT for developing IoT Applications using ESP-IDF framework.
> 
> Version: 3.0.0
> Build id: 20240605-1941
> 
> Copyright (c) 2024 Espressif Systems. All rights reserved.

kolipakakondal, 2024-08-30:
> Hi, I have reinstalled from scratch the Espressif IDE and it seems that after a fresh installation, the plugins installation from 'Install New Software...' is working.
> 
> After trying to install this plugin: [FreeRTOS Kernel Awareness for Eclipse from NXP](https://mcuoneclipse.com/2016/07/06/freertos-kernel-awareness-for-eclipse-from-nxp/), it fails and from now on, no more other Plugins can be installed. **(Note that in v2.8.1 this plugin can be installed without any problem)**.
> 
> Any thoughts about it?
> 
> As a workaround I have achieved to use this plugin in Espressif IDE v3.0.0 following these steps:
> 
> 1. First, install the plugin it in Espressif IDE v2.8.1 -> Installation succeeds
> 2. Import the plugin from Espressif IDE v2.8.1 to v3.0.0 as explained here:
>    
>    * https://stackoverflow.com/questions/11438529/easy-way-to-copy-old-plugins-into-new-install-of-eclipse
>    * https://stackoverflow.com/questions/3346333/exporting-installed-plugins-from-eclipse-to-another
> 
> This is a workaround, but, I would like to know:
> 
> 1. How to properly install the plugin ([FreeRTOS Kernel Awareness for Eclipse from NXP](https://mcuoneclipse.com/2016/07/06/freertos-kernel-awareness-for-eclipse-from-nxp/)) in Espressif IDE v3.0.0
> 2. How to fix the Espressif IDE once it is failing in installing any plugin.
> 
> Thanks,
> 
> Carles

Hi @carlessole This could be due to the [FreeRTOS Kernel Awareness for Eclipse from NXP](https://mcuoneclipse.com/2016/07/06/freertos-kernel-awareness-for-eclipse-from-nxp/) is not supporting Eclipse CDT 2024-03 and higher versions. I was not unable to install this on Eclipse CDT 2024-06 also. Maybe for the same reason it's failing with the Espressif-IDE 3.0.0 as it's based on Eclipse 2024-03.

alirana01, 2024-08-30:
@carlessole as mentioned by @kolipakakondal this plugin is not compatible with latest Eclipse CDT. Also I would advise not to use it with ESP-IDF since that plugin is based on an older version of FreeRTOS to fetch the data the ESP-IDF is using a more recent version so it probably wont be reporting anything in that context.

carlessole, 2024-08-31:
Thanks @kolipakakondal and @alirana01

Ok, it is probably not compatible as it is quite old, but I managed to install it importing it from an older Espressif IDE  (v2.8.1) installation. So, somehow it can be installed on Espressif-IDE 3.0.0 (or Eclipse 2024-03).

Then, do you know how to fix the Eclipse? As, after trying to install this plugin, I'm unable to install any other plugin. Something is broken and compatible plugins can not be installed anymore. For now, my solution has been to reinstall de whole Espressif IDE from scratch.

I also understand @alirana01, thanks, I will not use this plugin as it is quite old .
Have you got any plugin or other tool recommendation to use as FreeRTOS Task Awareness for better debugging my projects?

I finally decided to use the one embedded in latest MCUXpresso IDE:
-  In Espressif IDE 3.0.0 => File > Import > Install > From existing install, pick your MCUXpresso [C:\nxp\MCUXpressoIDE_11.9.0_2144\ide] , select the plug-in [MCUXpresso IDE FreeRTOS Task Aware Debug] and there you go!

And it is working fine:
[image]

[image]

[image]

Note that the plugin is not aware of the Dual Core, the Runtime total percentage is 200%, but it is not a problem.

Thanks,

Carles

alirana01, 2024-09-02:
@carlessole Thank you for the detailed response and feedback. At the moment we are not supporting any FreeRTOS task related plugins via our own IDE. If you need more information related to FreeRTOS tasks I can recommend that you can try to use the Debugger Console to execute GDB commands to fetch the required information directly from FreeRTOS's exposed variables. That although may require you to halt the application's execution