espressif/idf-eclipse-plugin #1043: ESP IDE 3.0.0 Disable in place editor content assist

C-basstien, 2024-08-30:
Dear ESP team,

I have just tried your new ESP-IDE 3.0.0: https://github.com/espressif/idf-installer/releases/tag/espressif-ide-3.0.0-esp-idf-5.2.2
The new C/C++ editor setting activate a content assist on functions parameters, where it show function argument name.
I highlighted in yellow in the following pictures, the arguments hint that the editor display:
[image]
How to disable this option ?

Thank you

kolipakakondal, 2024-08-30:
Hi @C-basstien Add the following settings to .clangd file.

```
InlayHints:
 Enabled: No
```

[image]

C-basstien, 2024-08-30:
Thanks a lot that worked.

closed 2024-08-30 by C-basstien