espressif/idf-eclipse-plugin #1047: org/eclipse/cdt/lsp/LspUtils error on clean installation

cjdg, 2024-09-13:
**Describe the bug**
When building a project it has the following error:

**To Reproduce**
Build project

**Expected behavior**
To dont have any error 

**Screenshots**
[image].

**Espressif-IDE Product Information:**

[IDF-info.txt](https://github.com/user-attachments/files/16986914/IDF-info.txt)

**Eclipse Error log:** 

[IDF-error.log](https://github.com/user-attachments/files/16986916/IDF-error.log)

kolipakakondal, 2024-09-13:
Hi @cjdg  Have you updated you eclipse version to the latest? I see some plugins are pointing to Eclipse 2024-09 version.

Here is our 3.1.0 update site artifacts which is not released yet, can you please check and see if that helps.
https://github.com/espressif/idf-eclipse-plugin/actions/runs/10663537001/artifacts/1881285738

You can download using the above link and update in your eclipse

cjdg, 2024-09-13:
It works!

DrStone71, 2024-09-18:
It works ! (Eclipse 2024-09 on Ubuntu and ESP-IDF 5.3.1)

Thanks !!!

DrStone71

closed 2024-09-26 by alirana01